artisan-roaster-scope/artisan #498: Artisan 2.4 Comparator profiles will not load.

dalmoody, 2020-08-20:
Mac Pro Mini, Sierra, BC-2MD roaster.
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/5104215/artisan-settings.txt)

Upon selecting files in Comparator; graphs should be generated as described in the instruction manual.

Files to be compared can be selected.  They are given a color code and name of file is displayed under title.  Files can be added and deleted.  No comparison graph is generated.

I have posted this question on Home Barista Forums; Roasting.

Here are the files I am trying to compare.
[Costa Rica La Minita Tarrazu_20-08-03_1258.txt](https://github.com/artisan-roaster-scope/artisan/files/5104139/Costa.Rica.La.Minita.Tarrazu_20-08-03_1258.txt)
[Costa Rica La Minita Tarrazu_20-08-17_1321.txt](https://github.com/artisan-roaster-scope/artisan/files/5104140/Costa.Rica.La.Minita.Tarrazu_20-08-17_1321.txt)

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2020-08-20:
Hi Dal,
Can you please confirm that you have at least BT checked under the top pull down?  Your profiles work fine for me here.  The screenshot shows them in the Comparator with BT, deltaBT, and Events checked.  Thanks.
Dave

[image]

MAKOMO, 2020-08-20:
Hm. Works for me as well. I did a factory reset on the official v2.4 macOS build, loaded your settings, started the comparator and "Add"ed your two profiles. Are you sure you sent us the Artisan settings file and the profiles you tested with?

[image]

dalmoody, 2020-08-20:
It works for me as well.  I’m very sorry for the inquest/issue.  Yes, Dave I checked the proper boxes, BT delta BT, events, and burner.  Align at charge.  After making a profile selection with the add button, I expected the comparison graph to appear in place of the selection window.  Instead the graph appears behind the window.  When I minimized the selection window, the graph appeared.

I’m old, so I use Artisan with a Mac Pro Mini and a large screen.  Artisan profiles a roast for me in the full screen mode.  It’s terrific.  That’s how the comparator issue occurred for me.

Again, I am very sorry if I have caused any alarms.  Artisan is a terrific roasting software.  You have been a great help.

Dal Moody

MAKOMO, 2020-08-20:
Thanks anyhow for your report. Happy roasting!

closed 2020-08-20 by MAKOMO