artisan-roaster-scope/artisan #504: On multiple event button Artisan Closes

reizao, 2020-09-15:
[configurações-artisan-240-crash.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/5224483/configuracoes-artisan-240-crash.aset.txt)

## Expected Behavior
Perform multiple event button

## Actual Behavior
When I press any multiple event button, the software crashes and close

## Steps to Reproduce the Problem

  1. Start a roast
  2. Click on any multiple event button
  3. Program Closes

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build (number in brackets shown in the about box): 164de804
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Phidgets 1046 + Probatone 5
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-09-25:
Thanks for reporting. Should be fixed in the latest continuous builds: https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

reizao, 2020-10-08:
Still happening with the new version.

MAKOMO, 2020-10-11:
@reizao : could you please test the current continuous builds to see if my last change fixes this issue for you and report back? THANKS!

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

reizao, 2020-10-11:
Dear Marko,

Works nice!!!

MAKOMO, 2020-10-11:
Thanks for reporting, testing and this positive feedback.

closed 2020-10-11 by MAKOMO