artisan-roaster-scope/artisan #505: fail to click OK on the Roast Properties page

Jimzgao, 2020-09-18:
## Expected Behavior
click OK on the Roast Properties page, then it should pass ok

## Actual Behavior
it didn't pass

## Steps to Reproduce the Problem

  1. input information of beans and roasting，click on OK button
  2. failed to next step
  3.I close the  the Roast Properties page
  4. save as an Excel file and an alog file
  5. the Excel file shows data, but the alog file shows nothing.
  6. also the excel file didn't show ohter markdowns except the original ones like FCs

## Specifications
  error log： Exception: deserialize() malformed node or string: @line 23653
  - Artisan Version:2.4.0
  - Artisan Build (number in brackets shown in the about box): Artisan 2.4.0 build 1 (164de804
[roast properties error.zip](https://github.com/artisan-roaster-scope/artisan/files/5243171/roast.properties.error.zip)
)
  - Platform (Mac/Windows/Linux + OS version):win10 1903    18362.1082
  - Connected devices or roasting machine: 
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-09-18:
Thanks for your report. It is not completely clear what happend, but the attached .alog file is defect. It contains NaN (not-a-number) elements as annotation coordinates which fails to load properly (the deserialize() malformed node error). Not sure why those NaN values for coordinates occur, however, I added code to prevent storing them in .alog profiles such that their reload cannot fail. Note that this does not fix the original source of the issue and it remains unclear to me if this fixes your issues with the Roast Properties dialog at all.

Could you share the steps on how you generated that .alog profile? A fixed version is attached below.

[20-09-17_2235-fixed.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/5245603/20-09-17_2235-fixed.alog.txt)

MAKOMO, 2020-09-18:
New builds are here: https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Jimzgao, 2020-09-18:
I appreciate your help very much. I will try this new version later and give feeback to you. You make the world of coffee roasting great！ Love u guys！

closed 2020-09-25 by MAKOMO