artisan-roaster-scope/artisan #506: Transposer Bug in Artisan 2.4.0 for Mac

drgarys, 2020-09-19:
## Expected Behavior: Insert new target drop temp and have curve adjusted.

## Actual Behavior: Error

## Steps to Reproduce the Problem

  1. Load profile
  2. Start Transposer
  3. Insert new target drop temp
4. Select OK or Apply. 
5. Either choice returns error code.

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build (number in brackets shown in the about box):5afdae1
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.15.6
  - Connected devices or roasting machine: Phidget 1048 4xTC 01
[image]
[image]
[TransposerErrorReport091920.txt](https://github.com/artisan-roaster-scope/artisan/files/5250392/TransposerErrorReport091920.txt)
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/5250396/artisan-settings.aset.txt)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

drgarys, 2020-09-19:
Here's the alog file I was using.
[EthiopiaDPExtendedMaillardDesignedProfile091920_20-09-19_1100.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/5250405/EthiopiaDPExtendedMaillardDesignedProfile091920_20-09-19_1100.alog.txt)

MAKOMO, 2020-09-19:
Thanks for your detailed report. I just fixed this border case. A new continuous build containing the fix should be available in some minutes.

closed 2020-09-19 by MAKOMO

drgarys, 2020-09-19:
Thanks, Marko!

drgarys, 2020-09-19:
I just tried to download again and got the same release that has the bug. Please let me know when you have the fixed version available for download.

roasterdave, 2020-09-19:
Hi Gary, be sure you are grabbing the continuous build.  The version will show as 2.4.1.  It is available here: 
https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

drgarys, 2020-09-19:
Dave,

I just downloaded it, but my Mac won't open it. I get an error message that the Artisan.app can't be opened because the developer can't be verified. MacOS cannot verify that this app is free from malware.

MAKOMO, 2020-09-20:
Gary, the continous builds generated automatically by our CI systems are NOT developer signed. Thus you cannot start them as usual by a simple double-click. You need to do a right-click on the app icon and select OPEN. macOS will then ask you if you trust that application and if you confirm it will start the app. Note that this specific way of starting the app is only required on the first run of unsigned apps. From the first successful start on you can start that app as usual. This is the Apple way to protect you from bad things happening...

drgarys, 2020-09-20:
Thank you Marko. That worked, and the fix works. For anyone else viewing this, the screen shots follow:

[image]
[image]

Thank you and Dave for showing me how to use the continuous development version to access a bug fix and how to open it on my Mac. 

I'm really enjoying the Analyzer and Transposer features for creating and adjusting background roast curves that also guide bean temperature rate of rise (BT ROR).