artisan-roaster-scope/artisan #507: Quick Load Background in Roast Properties Dialog

stinsonddog, 2020-09-25:
## Expected Behavior
Roast Properties should contain the sole place to prepare to roast a particular bean/batch.  Currently you have do multiple steps to load a background roast.  

## Actual Behavior

Once your Roast Properties is set you have to 
Close Roast Properties
Open Roast Background
Load the Roast
Close Roast Background

Shortcuts make it easier, yet for the new user and experienced user there should be only one dialog to input the info for a roast.  The current setup can be left in place for advanced use of the background function.  So basically there would be two places to load a background.  On the Roast Properties page its really "Quick Load Background". 

Hopefully more people will use backgrounds. Also using Analyzer makes it necessary to use this more often and the greater need for it.  After using Analzyer the background needs to be reloaded at times.  

It's a workflow issue.  

There is space to add the Quick Load line at the bottom of Roast Properties.  Images are from a 13" Mac.  If not the second image is a start to adjust spacing. 

  - Artisan Version: 2.4
 
[image]

Move Density and Moisture over.  Adjust where Calc goes and you pick up two lines.  
[image]

MAKOMO, 2020-09-26:
Mine looks like this:

[image]

Quite crowded already. Why not use the Recent Roast feature to load a background profile with one click in this dialoog and even set the batch size along?

InvisibleOrdinary, 2020-09-26:
I second this request, even though this page is pretty crowded.
Consider that when running through a series of batches that contains more than one coffee, you often need to make this update while the batch is already started. The Recent Roast, or the saved roasts dropdown from the "title' are good (are they the same thing?) so why not be able to attach the profile to that, so when you select the batch, it also loads the correct profile. It would be automatic, and errors would really only be if it can't find the profile.
Of course you need a place to select the correct profile in the first place, but it's an important tool when roasting, so seems finding a place is worth it.

stinsonddog, 2020-09-26:
Ok yours looks more crowded.
Your suggestion is great for many instances especially a similar bean, but often there is a need to look and find other roasts for backgrounds.  Simple example is trying something different like a light roast and dark roast to blend. 

So two suggestions
Why can't it be like Recent Roast at the bottom with a + or - to choose it so it only takes up one line?  OR
Why does Beans need 4-5 lines.  Why not just take 2 from there and seems to be plenty.

MAKOMO, 2020-09-26:
> The Recent Roast, or the saved roasts dropdown from the "title' are good (are they the same thing?) so why not be able to attach the profile to that, so when you select the batch, it also loads the correct profile

Adding a roast to the Recent Roast list does load it in the background on selecting the generated entry. So this behavior is exactly what you are asking for, right?

InvisibleOrdinary, 2020-09-26:
Where do you choose the background profile? When I change coffees in the middle of a session, I always also have to choose the correct background...

stinsonddog, 2020-09-26:
@InvisibleOrdinary You choose it under Roast>Background, bottom of dialog box.  This is what I am asking for a Quick Load in the Properties dialog.  If you save a profile to Recent Roast as Marko suggested and there is a background profile, it will load that background when you choose the same bean to roast in the future.  That helps but doesn't cover all the scenarios.  For example you did an improved roast since the one you saved and you now wish to use that one.

[image]

InvisibleOrdinary, 2020-09-27:
Sorry, I wasn't clear in my question… yeah, that is where you load the roast - but I meant to ask, where does it get attached to the 'Recent Roast' profile in Roast Properties?  from what I see, it doesn't attach any background profile to that… so I still don't understand Marko's reply.

MAKOMO, 2020-09-27:
` If you save a profile to Recent Roast as Marko suggested and there is a background profile, it will load that background when you choose the same bean to roast in the future.`

That is not fully correct. If you add a Recent Roast the current loaded profile (not its background) is associated with this new entry. On selecting that entry, that profile is loaded in the background. The profile is attached "internally". This is not visible in any UI, but on selecting that entry it loads that associated profile into the background.

Adding a Recent Roast entry with the name and batch size of an already existing one, just updates the existing one (incl. the profile association) instead of adding a new one covering your scenario. 

Anyhow, you can at any moment press the h key to quickly load a different background profile or press ALT-H to remove the currently loaded profile.

stinsonddog, 2020-09-27:
I had not used Recent Roasts that way.  I think you can close this and the shortcuts are fine for now.

closed 2020-09-28 by MAKOMO