artisan-roaster-scope/artisan #508: Artisan Will Not Important Aillio Bullet .JSON Files downloaded from Roast World

jevenator, 2020-10-01:
Hi all, 

I have an Aillio Bullet that I'd like to import into Artisan. I downloaded a fresh copy of 2.4.0 and the only way to download roasts is from https://roast.world/my/roasts as an export. It saves it as a .json file.

I get this message 

> Exception: self.importBullet() 'sampleRate'@line 31716

and nothing shows up on the screen. Tried it for two different exports.
[Brazil Bob-o-Link.txt](https://github.com/artisan-roaster-scope/artisan/files/5309573/Brazil.Bob-o-Link.txt)

I have uploaded the .JSON file that I renamed as a .txt.

MAKOMO, 2020-10-01:
Aillio obviously changed the file format. I patched the import and the import should now work. Could you please verify this by downloading and installing the latest beta build from: https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

[image]

jevenator, 2020-10-01:
Yes, it works now...How do I get the same annotations out as you have it. Everything shows except all the labels for power and air and drum speed.

Another thing that I'm noticing. In the Bullet R1 V2 There is no real ET probe that is being recorded. The BT probe is an actual BT probe that is in the machine but the temperature that it records is not really a reliable source of information during the roast for smaller batch sizes which is what I did here (350g). Comparing other roasts from people, 1kg charges do have an accurate BT reading. So for smaller roasts what I'm doing it's relying on the IBTS sensor which is accurate here. As a result, when I convert to F in Artisan it provides very small numbers based on however it is converting those values. On the roast world it will provide only C but on their RoasTime software it shows it as BT & IBTS and will reliably convert it between C and F

MAKOMO, 2020-10-02:
Please check the [documentation](https://artisan-scope.org/docs/events/) and the various blog post on Events and how to adjust their renderings.

The Aillio importer was not respecting the current temperature unit. I fixed this in the just released v2.4.2. To convert a profiles temperature curves to a different temperature unit you use the entry "Convert Temperature Profile" under the Tools menu.

closed 2020-10-02 by MAKOMO