artisan-roaster-scope/artisan #510: Ubuntu 20.04 desktop artisan 2.4.2 does not launch after installation

larrym, 2020-10-05:
## Expected Behavior
Launch Artisan

## Actual Behavior
Nothing - 

## Steps to Reproduce the Problem

  1. Download artisan-linux-2.4.2.deb from github
  2. sudo apt install ./artisan-linux-2.4.2.deb
$ sudo apt install ./artisan-linux-2.4.2.deb 
[sudo] password for larry: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'artisan' instead of './artisan-linux-2.4.2.deb'
The following NEW packages will be installed:
  artisan
0 upgraded, 1 newly installed, 0 to remove and 3 not upgraded.
Need to get 0 B/159 MB of archives.
After this operation, 395 MB of additional disk space will be used.
Get:1 /home/larry/Downloads/artisan-linux-2.4.2.deb artisan amd64 2.4.2 [159 MB]
Selecting previously unselected package artisan.
(Reading database ... 215044 files and directories currently installed.)
Preparing to unpack .../artisan-linux-2.4.2.deb ...
Unpacking artisan (2.4.2) ...
Setting up artisan (2.4.2) ...
Processing triggers for man-db (2.9.1-1) ...
Processing triggers for desktop-file-utils (0.24-1ubuntu3) ...
Processing triggers for mime-support (3.64ubuntu1) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu1) ...

  3. Attempt to launch Artisan does nothing

## Specifications

  - Artisan Version: 2.4.2
  - Artisan Build (number in brackets shown in the about box): ?
  - Platform (Mac/Windows/Linux + OS version): Ubuntu 20.04 Desktop
  - Connected devices or roasting machine: ?
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-10-05:
Larry. Thanks for this detailed report. It turns out that this is a build bug of the linux builds. This just got fixed and new builds have been uploaded. Could you please re-download and re-install and report if this fixes your issue?

larrym, 2020-10-05:
Thank you, it works.

[image]

MAKOMO, 2020-10-05:
Excellent!

closed 2020-10-05 by MAKOMO