artisan-roaster-scope/artisan #513: Question about RoR calculation

rmongiovi, 2020-10-13:
I've noticed something curious with Artisan RoR on my hottop.  I press the On button in Artisan to start warming up the roaster and I have the LCDs displayed on the right hand side of the screen (since I needed them to turn "super" on).  As the roaster warms up I see the RoR increase until it gets to the 50-60F range.  When ET gets to 250F I hit Start to begin recording.

I noticed that immediately after hitting Start the ET and BT RoR numbers in the LCDs drop to 0.  They rise over the next 50 seconds or so until they're back in the 50-60F range.  Since the hottop is electric and definitely doesn't change temperature that quickly this must have to do with Artisan averaging values over a 50 second interval.  But where does that number come from?

I have my sampling interval set to 2 seconds and oversampling turned on.  The curve filter has smooth curves set to 5.  The RoR filter has delta span set to 1s and smoothing set to 3.

I first noticed this behavior quite some time ago, but I just upgraded to 2.4.2 hoping that the RoR computation improvements might address it.  I still see the same 50 second delay before RoR gets back to where it should be.  It seems to me that a more quickly responsive RoR curve would be a good thing (assuming it doesn't get too noisy).  Is the 50 seconds I see baked in or somehow a product of my smoothing settings?

FilePhil, 2020-12-17:
When you drop in the Beans, the ROR will not only drop to zero, but way lower. The Beans taking the heat "away" and the temperature drops immediately in a short time. This means, the Temperature Change per Time( ROR) is below zero/near zero before the Turing Point. 
At the Turning Point the heat is increasing and the BT ROR is becoming positive again. This is causing the "big" delay and the sharp curve at the beginning of the BT ROR.

Maybe this Cite from a Blogpost helps:
>  RoR lags in time - always!
>   BT RoR is by definition zero at the turn around point (TP)
>   the time lag produced by computing the RoR is half of the delta-span interval
>  the BT RoR time lag is the difference between that TP and where the computed RoR gets zero
>   the RoR acts stronger, but never earlier to temperature changes than the bean curve as is sometimes stated
>   the underlying temperature signal is also shifted by unavoidable measuring delays, resulting in a further shifts of the RoR w.r.t. to the real bean temperatures
[https://artisan-roasterscope.blogspot.com/2019/03/on-idle-noise.html](https://artisan-roasterscope.blogspot.com/2019/03/on-idle-noise.html)

Further Information: 
[https://artisan-roasterscope.blogspot.com/2014/01/sampling-interval-smoothing-and-rate-of.html](https://artisan-roasterscope.blogspot.com/2014/01/sampling-interval-smoothing-and-rate-of.html)
[https://www.cropster.com/news/article/focus-onthe-rate-of-rise/](https://www.cropster.com/news/article/focus-onthe-rate-of-rise/)

rmongiovi, 2020-12-17:
I think you misunderstood me.  I didn't add beans.  I just pressed the Start button.  Any RoR change is an artifact of the software with no physical cause.

The blogpost you quote is exactly my issue.  My delta-span is 1 second yet it takes 50 seconds for RoR to recover after pressing the Start button with no change in the roaster heat setting or beans added to the chamber.  It looks to me as though the RoR is being averaged over 50 seconds worth of samples regardless of my settings.  Which really calls into question how responsive it can be to changes in the roaster settings.

MAKOMO, 2021-01-12:
The behavior you report is an artefact caused by the buffer of readings used to compute the RoR during ON (just sampling but not yet recording) is cleared on starting a new recording with START. Once recording the RoR is computed from the data available in the recording only, thus initially starting from an empty set of data on pressing START. We are aware of this minor defect. Not sure this will be changed/improved in future versions due to the limited development resources available. Note that the RoR computation during ON was just introduced in v2.4.0 as a quick hack. So what we have now is better than nothing, but not perfect yet. I agree.

closed 2021-01-12 by MAKOMO