artisan-roaster-scope/artisan #514: Giesen machine stop connected S7 with artisan 2.4.2 after 6 minutes

kafekopteio, 2020-10-16:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.4.2
  - Artisan Build (number in brackets shown in the about box):(3d73fd95)
  - Platform (Mac/Windows/Linux + OS version):Windows-10
  - Connected devices or roasting machine:Giesen W1A & W60A
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-10-16:
Any entry in your error log (menu Help >> Errors)?
Could you please share also your Serial Log (tick the flag in menu Help >> Serial and record some minutes).
Thanks!

MAKOMO, 2020-10-17:
Found and fixed (Commit 9a2d2a1935ed0ee6fab276b71832e620a4c023ef) an issue in the Siemens S7 optimizer that resulted in more instead of less requests to the connected machine.

Could you please test if the v2.4.3 beta build here:
https://github.com/artisan-roaster-scope/artisan/releases/continuous
fixes your issue?

Thanks for your support,
Marko

kafekopteio, 2020-10-18:
Hello Marko, i used the Giesen WxA+ IR Env as machine but didn't work. My machines works with WxA settings.

Thanks in advance
Antonis

⁣Εστάλη από το BlueMail ​

MAKOMO, 2020-10-18:
Dear Antonis,

all those WxA+ settings require to register the Cropster Code into the Giesen machine, which can be purchased from Giesen. Without this code you need to configure using the regular WxA settings. Can you confirm that the issue you report is gone and the machine stays connected longer than 6min?

MAKOMO, 2020-10-18:
Antonis,

why did you choose that WxA+ IR Env setup? Do you have those optional IR and Env. Temp. sensor installed in your machine?

closed 2020-10-20 by kafekopteio

marcodegroot, 2020-10-26:
I've had the same issue with 2.4.2. But installing 2.4.3 beta gives other issues:

> 500 12:07:30.956 Error: automaticsave() [Errno 24] Too many open files: 'C:/Users/.....\\20-10-26_1152.alog'@line 22623
>...
> 294 12:06:16.221 S7 Communication Error readInt: Communication error
> 293 12:06:15.576 S7 Communication Resumed
> 292 12:06:15.349 S7 Communication Error writeInt: [Errno 24] Too many open files: 'nul'
> 291 12:06:14.237 S7 Communication Error readInt: Communication error
> 290 12:06:14.221 S7 Communication Error readInt: Communication error
> .....
> 7 12:04:33.551 S7 Communication Resumed
> 6 12:04:33.215 S7 Communication Error readInt: Communication error
> 5 12:04:33.197 S7 Communication Error readInt: Communication error
> 4 12:04:32.556 S7 Communication Resumed
> 3 12:04:32.209 S7 Communication Error readInt: Communication error
> 2 12:04:32.193 S7 Communication Error readInt: Communication error
> 1 12:04:31.574 S7 Communication Resumed
> 

Returning to 2.4 seems to solve everything for me.