artisan-roaster-scope/artisan #515: Slider value automatically changing

greencardigan, 2020-10-19:
## Expected Behavior
Slider value should remain on the set value until changed.

## Actual Behavior
Slider value changes without any user input.

## Steps to Reproduce the Problem

  1.  Set slider max to 20
  2.  Start logging
  3.  Change slider to 15
  4.  Wait for slider value to change to 3

Presumably the slider value is getting scaled based on the max slider value set. eg 15 * (20 / 100) = 3

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build (number in brackets shown in the about box): 5b99d6f
  - Platform (Mac/Windows/Linux + OS version): RPi Linux Stretch
  - Connected devices or roasting machine: TC4
  
[250g Roaster Settings2.txt](https://github.com/artisan-roaster-scope/artisan/files/5398987/250g.Roaster.Settings2.txt)

MAKOMO, 2020-10-20:
Your slider is moved by your quantifier definition (menu Config >> Events, 4th tab Quantifiers). Either deactivate that fan quantifier or set its range to correspond to the of the corresponding slider which is [0,20]. Does this resolves the issue?

greencardigan, 2020-10-20:
Thanks Marko. That was an easy fix! Sorry I had forgotten what the quantifiers did.

closed 2020-10-20 by greencardigan