artisan-roaster-scope/artisan #517: Axis and curve parameters not stored

Kortenbach, 2020-10-21:
## Expected Behavior
Any changes made in the Axis and curve parameters dialog should be reflected in the screen.

## Actual Behavior
The changes are visible after clicking apply BUT revert back to their old settings after clickin OK.

## Steps to Reproduce the Problem

  1. Install Artisan
  2. Click "Edit axis and curve parameters" button
  3. Change any setting and click "apply"
  4. Click "OK" and watch your settings disappear
  5. Re-open the "Edit axis and curve parameters" dialog
  6. Confirm that the settings changed back to their original values.

## Specifications

  - Artisan Version: 2.4.2.
  - Artisan Build (number in brackets shown in the about box): 3
  - Platform (Mac/Windows/Linux + OS version): Win 10
  - Connected devices or roasting machine: None
  
Fresh install

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-10-21:
This is a known issue and unlikely to be fixed. However, the data under Curves is persisted in that dialog and can be used to change the eg curves line style. Please use the Axis dialog to configure axis (menu Config >> Axis).

Kortenbach, 2020-10-21:
Thanks for the feedback. I will try ASAP. Meanwhile I tried 2.4.0 and I found that, when installingV 2.4.2., windows defender shows a warning whereas the V2.4.0 installer does not. Maybe this is helpfull information for you...

Kortenbach, 2020-10-21:
Using menu Config >> Axis partly fixed the problem. For instance: I could still not get the legend to show. But otherwise it seems to be an acceptable work-around. Thanks for the quick reply!

closed 2020-11-05 by MAKOMO