artisan-roaster-scope/artisan #518: dist folder not included in last distro

adroc, 2020-10-23:
## Expected Behavior

## Actual Behavior

/artisan-2.4.2/src$ ./build-rpi2-deb.sh 
+ python3 -c import artisanlib; print(artisanlib.__version__)
+ VERSION=2.4.3
+ NAME=artisan-linux-2.4.3
+ sed -i -e s/_VERSION_/2.4.3/g debian/DEBIAN/control
+ sed -i -e s/x86_64/armhf/g debian/DEBIAN/control
+ gzip -9 debian/usr/share/man/man1/artisan.1
+ chmod +r debian/usr/share/man/man1/artisan.1.gz
+ gzip -9 debian/usr/share/doc/artisan/changelog
+ chmod +r debian/usr/share/doc/artisan/changelog.gz
+ chmod +r debian/usr/share/applications/artisan.desktop
+ chmod -x debian/usr/share/applications/artisan.desktop
+ chmod +rx debian/usr/bin/artisan
+ chmod -R +r dist
chmod: cannot access 'dist': No such file or directory

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:  2.4.2
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):  ubuntu
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-10-23:
I cannot see this in any of the recent build logs, incl. the very last, for the RPi platform. Could you please point to the entry in the build history where you observed this?

MAKOMO, 2020-10-23:
The last build <https://travis-ci.org/github/artisan-roaster-scope/artisan/jobs/738109471> reads:

```
...
+ chmod +rx debian/usr/bin/artisan
+ chmod -R +r dist
+ chmod +x dist/icons
+ tar -cf dist-rpi.tar dis
...
```

adroc, 2020-10-23:
I encountered this when I tried to build from source. 2.4.2.  Originally I tried to install from 3 different .deb packages with no luck.

artisan-linux-2.4.2.deb                  
artisan-linux-2.4.2_raspbian-stretch.deb
artisan-linux-2.4.2_raspbian-buster.deb 

ubuntu@ubuntu:~/Downloads/artisan-2.4.2$ ls
CODE_OF_CONDUCT.md  ISSUE_TEMPLATE.md  LICENSE  README.md  doc  src  wiki
ubuntu@ubuntu:~/Downloads/artisan-2.4.2$ cd src

ubuntu@ubuntu:~/Downloads/artisan-2.4.2/src$ ls -l
total 4964
-rw-rw-r-- 1 ubuntu ubuntu   4201 Oct  2 17:32 Info.plist
-rw-rw-r-- 1 ubuntu ubuntu   3725 Oct  2 17:32 README.txt
drwxrwxr-x 5 ubuntu ubuntu   4096 Oct  2 17:32 Wheels
-rw-rw-r-- 1 ubuntu ubuntu    241 Oct  2 17:32 artisan-alog.xml
-rw-rw-r-- 1 ubuntu ubuntu    239 Oct  2 17:32 artisan-alrm.xml
-rw-rw-r-- 1 ubuntu ubuntu    241 Oct  2 17:32 artisan-apal.xml
-rw-rw-r-- 1 ubuntu ubuntu    241 Oct  2 17:32 artisan-aset.xml
-rw-rw-r-- 1 ubuntu ubuntu    239 Oct  2 17:32 artisan-athm.xml
-rw-rw-r-- 1 ubuntu ubuntu   1410 Oct  2 17:32 artisan-linux.spec
-rw-rw-r-- 1 ubuntu ubuntu   2483 Oct  2 17:32 artisan-mac.spec
-rw-rw-r-- 1 ubuntu ubuntu    234 Oct  2 17:32 artisan-wg.xml
-rwxr-xr-x 1 ubuntu ubuntu   4893 Oct  2 17:32 artisan-win.spec
-rw-rw-r-- 1 ubuntu ubuntu 217396 Oct  2 17:32 artisan.icns
-rw-rw-r-- 1 ubuntu ubuntu 110081 Oct  2 17:32 artisan.ico
-rw-rw-r-- 1 ubuntu ubuntu  66864 Oct  2 17:32 artisan.png
-rw-rw-r-- 1 ubuntu ubuntu   2832 Oct  2 17:32 artisan.pro
-rw-rw-r-- 1 ubuntu ubuntu   3519 Oct  2 17:32 artisan.py
-rw-rw-r-- 1 ubuntu ubuntu 429402 Oct  2 17:32 artisanAlarms.icns
-rwxr-xr-x 1 ubuntu ubuntu 370070 Oct  2 17:32 artisanAlarms.ico
-rw-rw-r-- 1 ubuntu ubuntu 335093 Oct  2 17:32 artisanPalettes.icns
-rwxr-xr-x 1 ubuntu ubuntu 370070 Oct  2 17:32 artisanPalettes.ico
-rw-rw-r-- 1 ubuntu ubuntu 406504 Oct  2 17:32 artisanProfile.icns
-rwxr-xr-x 1 ubuntu ubuntu 370070 Oct  2 17:32 artisanProfile.ico
-rw-rw-r-- 1 ubuntu ubuntu 302357 Oct  2 17:32 artisanSettings.icns
-rw-rw-r-- 1 ubuntu ubuntu 370070 Oct  2 17:32 artisanSettings.ico
-rw-rw-r-- 1 ubuntu ubuntu 274815 Oct  2 17:32 artisanTheme.icns
-rw-rw-r-- 1 ubuntu ubuntu 370070 Oct  2 17:32 artisanTheme.ico
-rw-rw-r-- 1 ubuntu ubuntu 313375 Oct  2 17:32 artisanWheel.icns
-rw-rw-r-- 1 ubuntu ubuntu 370070 Oct  2 17:32 artisanWheel.ico
drwxrwxr-x 3 ubuntu ubuntu   4096 Oct 23 01:09 artisanlib
-rwxr-xr-x 1 ubuntu ubuntu   2412 Oct  2 17:32 build-linux-pkg.sh
-rwxr-xr-x 1 ubuntu ubuntu   3718 Oct  2 17:32 build-linux.sh
-rwxr-xr-x 1 ubuntu ubuntu   5991 Oct  2 17:32 build-mac3-pi.sh
-rwxr-xr-x 1 ubuntu ubuntu   2462 Oct  2 17:32 build-mac3.sh
-rwxr-xr-x 1 ubuntu ubuntu   1291 Oct  2 17:32 build-rpi2-deb.sh
-rwxr-xr-x 1 ubuntu ubuntu   4075 Oct  2 17:32 build-rpi2-pi.sh
-rwxr-xr-x 1 ubuntu ubuntu   5552 Oct  2 17:32 build-rpi2-qemu.sh
-rwxr-xr-x 1 ubuntu ubuntu    640 Oct  2 17:32 build-win3-pi.bat
drwxrwxr-x 2 ubuntu ubuntu   4096 Oct  2 17:32 comm
drwxrwxr-x 2 ubuntu ubuntu   4096 Oct  2 17:32 const
-rw-rw-r-- 1 ubuntu ubuntu   1039 Oct  2 17:32 create_dmg.py

drwxrwxr-x 5 ubuntu ubuntu   4096 Oct  2 17:32 debian
## I believe if dist/ was included in the source tar/zip it would be here.

drwxrwxr-x 2 ubuntu ubuntu   4096 Oct  2 17:32 help
drwxrwxr-x 2 ubuntu ubuntu   4096 Oct  2 17:32 icons
drwxrwxr-x 5 ubuntu ubuntu   4096 Oct  2 17:32 includes
drwxrwxr-x 2 ubuntu ubuntu   4096 Oct  2 17:32 misc
drwxrwxr-x 2 ubuntu ubuntu   4096 Oct  2 17:32 plus
-rw-rw-r-- 1 ubuntu ubuntu     26 Oct  2 17:32 qt-win.conf
-rw-rw-r-- 1 ubuntu ubuntu     26 Oct  2 17:32 qt.conf
drwxrwxr-x 3 ubuntu ubuntu   4096 Oct  2 17:32 raspbian
-rwxr-xr-x 1 ubuntu ubuntu    423 Oct  2 17:32 requirements-linux.txt
-rwxr-xr-x 1 ubuntu ubuntu    690 Oct  2 17:32 requirements-osx.txt
-rwxr-xr-x 1 ubuntu ubuntu   1062 Oct  2 17:32 requirements-rpi.txt
-rwxr-xr-x 1 ubuntu ubuntu    570 Oct  2 17:32 requirements-win.txt
-rwxr-xr-x 1 ubuntu ubuntu    380 Oct  2 17:32 requirements.txt
-rw-rw-r-- 1 ubuntu ubuntu  13487 Oct  2 17:32 setup-install.nsi
-rwxr-xr-x 1 ubuntu ubuntu  12472 Oct  2 17:32 setup-install3-pi.nsi
-rwxr-xr-x 1 ubuntu ubuntu  14035 Oct  2 17:32 setup-install3.nsi
-rw-rw-r-- 1 ubuntu ubuntu  13731 Oct  2 17:32 setup-mac.py
-rw-rw-r-- 1 ubuntu ubuntu  19640 Oct  2 17:32 setup-mac3.py
-rw-rw-r-- 1 ubuntu ubuntu   5366 Oct  2 17:32 setup-packages-centos7.txt
-rw-rw-r-- 1 ubuntu ubuntu  11159 Oct  2 17:32 setup-packages-linux.txt
-rw-rw-r-- 1 ubuntu ubuntu  23555 Oct  2 17:32 setup-packages-mac.txt
-rw-rw-r-- 1 ubuntu ubuntu   3632 Oct  2 17:32 setup-packages-raspberry.txt
-rw-rw-r-- 1 ubuntu ubuntu   6544 Oct  2 17:32 setup-packages-windows.txt
-rw-rw-r-- 1 ubuntu ubuntu   5107 Oct  2 17:32 setup-packages-windows3.txt
-rw-rw-r-- 1 ubuntu ubuntu   4457 Oct  2 17:32 setup-packages-windows35.txt
-rwxr-xr-x 1 ubuntu ubuntu   7374 Oct  2 17:32 setup-win.py
-rwxr-xr-x 1 ubuntu ubuntu   7792 Oct  2 17:32 setup-win3.py
-rw-rw-r-- 1 ubuntu ubuntu   7430 Oct  2 17:32 setup-win35.py
-rw-rw-r-- 1 ubuntu ubuntu   2115 Oct  2 17:32 setup.nsi
drwxrwxr-x 2 ubuntu ubuntu   4096 Oct  2 17:32 translations

MAKOMO, 2020-10-23:
Building Artisan from source is non-trivial and require a specific setup. The dist folder is not part of the Artisan source as it is the part of the actual build and generated by the scripts.

Note that the "artisan-linux-2.4.2.deb" build is for Linux running on Intel processors. The other two Linus builds marked in their file name with "raspbian" are made for the RPi running Raspbian on Arm processors. They should work on standard configuration. It would be interesting to see why those do not work for you. First you should install the one of those two based on your version of Raspbian. For now Stretch and Buster are supported. After installation you could check the messages you receive by starting Artisan via

```
# /usr/share/artisan/artisan
```

Please paste the output here and I will take a look.

adroc, 2020-10-24:
I reinstalled the pi4 with a raspbian instead of ubuntu. 
I historically didn't have a problem installing from source and semi needed to facilitate using odroid-xu4 platform.  (early versions of artisan.)

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

pi@raspberrypi:~ $ /usr/share/artisan/artisan
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.
Aborted

MAKOMO, 2020-10-25:
Could you try to start via

```
pi@raspberrypi:~ $ exec /usr/bin/env PATH=/usr/share/artisan:$PATH QT_PLUGIN_PATH=/usr/share/artisan/PyQt5/Qt/plugins LD_LIBRARY_PATH=/usr/share/artisan:$LD_LIBRARY_PATH /usr/share/artisan/artisan
```

obviously you do not have the Qt plugins in your path.

adroc, 2020-10-27:
exec /usr/bin/env PATH=/usr/share/artisan:$PATH QT_PLUGIN_PATH=/usr/share/artisan/PyQt5/Qt/plugins LD_LIBRARY_PATH=/usr/share/artisan:$LD_LIBRARY_PATH /usr/share/artisan/artisan &
[1] 16580
pi@raspberrypi:~ $ Error in sys.excepthook:
Traceback (most recent call last):
  File "artisanlib/main.py", line 34947, in excepthook
AttributeError: 'NoneType' object has no attribute 'qmc'

Original exception was:
Traceback (most recent call last):
  File "artisan.py", line 90, in <module>
  File "artisanlib/main.py", line 34999, in main
  File "artisanlib/main.py", line 14316, in __init__
  File "artisanlib/main.py", line 1489, in __init__
AttributeError: 'QDateTime' object has no attribute 'toSecsSinceEpoch'
[16581] Failed to execute script artisan

MAKOMO, 2020-11-24:
fixed for v2.4.4

closed 2020-11-24 by MAKOMO