artisan-roaster-scope/artisan #519: Windows Defender reports trojan virus in V2.4.2

Kortenbach, 2020-10-27:
## Expected Behavior
No Virus

## Actual Behavior
Virus detected and Artisan.exe removed by windows Defender.

## Steps to Reproduce the Problem

  1. Install Artisan
  2. Scan Artisan.exe with Windows Defender
  3. 

## Specifications

  - Artisan Version: 2.4.2
  - Artisan Build (Sorry, now in quarantine):
  - Platform Win10
  - Connected devices or roasting machine: n/a
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

zenhiker, 2020-10-28:
Seeing the same issue when installing. As a workaround, you can allow the file to run on the device, but that may not be the best permanent solution. Threat detected is "Trojan:Win32/Zpevdo.B" 

Affected items:
file: C:\Program Files\Artisan\artisan.exe
file: C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Artisan\Artisan.lnk
file: C:\Users\Public\Desktop\Artisan.lnk
regkey: HKLM\SOFTWARE\Wow6432Node\MICROSOFT\WINDOWS\CURRENTVERSION\UNINSTALL\Artisan
startup: C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Artisan\Artisan.lnk
uninstall: HKLM\SOFTWARE\Wow6432Node\MICROSOFT\WINDOWS\CURRENTVERSION\UNINSTALL\Artisan

MAKOMO, 2020-11-24:
v2.4.4 improves on this by compiling its own pyinstaller bootloader

closed 2020-11-24 by MAKOMO