artisan-roaster-scope/artisan #521: No temp readings

redadder22, 2020-11-05:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

redadder22, 2020-11-05:
Artisan 2.1.2 (e5e2c5d) 
Mac
Phidgets 1048

Other problems: won't quit when I try to shut down the computer, and won't switch off after roasting.

MAKOMO, 2020-11-24:
Looks like a configuration or installation issue and not a issue in the software. Please ask for support on those on the Artisan mailing list or a public forum.

closed 2020-11-24 by MAKOMO