artisan-roaster-scope/artisan #523: Errno24: Too many open files

shokunincoffee, 2020-11-19:
## Expected Behavior
After dropping a batch, we want to finish the roast and (auto)save it to our computer.
Also, we want our event annotations to show during a roast and the Gas events to all show along a curve instead of on one horizontal line.

## Actual Behavior
All Gas events are registered at the same Y-value and don't show annotations until we end the roast. Then, these annotations all overlap (because of the same Y-value).

Upon ending the roast and trying to save it, we get the error code below. However, when we try to replicate this issue, it does save normally (I guess this is another case of when talking to tech support, everything works again).

500 12:37:45.560 Error: automaticsave() [Errno 24] Too many open files: 'C:/Users/jelle/OneDrive/Documents/Artisan\\4 El Sopapo 2020-11-18_1227.alog'@line 22605

## Steps to Reproduce the Problem

  1. Start a roast.
  2. Change Gas setting a few times.
[Artisan.zip](https://github.com/artisan-roaster-scope/artisan/files/5566629/Artisan.zip)

  3. End it after more than 10 minutes (to mimick a normal roast)
  

## Specifications

  - Artisan Version: 2.4.3 (also tried on 2.4.2)
  - Artisan Build (number in brackets shown in the about box): 138dd993
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Giesen W6A+ (using Siemens S7)
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-11-19:
Dear Jelle,

thanks for opening this issue. We will investigate. One more question: is there anything in the error log of Artisan (menu Help >> Errors). I am asking as it might be that there is a problem with the connection (or setup) which might lead to communication errors and collecting too many of those might lead to this "too many open files" issue. Just speculating..

MAKOMO, 2020-11-19:
One thing in you setup is not "optimal". You defined your buttons "relative" using the "+-Burner" which will add the given value to the current burner value. You might want to use the event type "Burner" without the "+-" instead, or use the "+-Burner" event type to have "+10%" and "-10%" buttons with an S7 command setDBint(2,14,{}+1). This should resolve your gas event issue. The events generated by the custom defined buttons as you defined them always evaluate to 100 (same Y-values).

I see further that the Burner PV reading that Artisan receives is always 0. It seems that on your machine this value can not properly retrieved from the machine for some reason (different software!?). It would be interesting to check this Artisan log for any communication problems or even the Artisan serial log (menu Help >> Serial) could shed some light on this.

The too many open files issue might also resolve itself by rebooting the PC as you might have hit a limit (memory or otherwise) for some reasons not directly related to Artisan on this one.

I see that you have oversampling and autosave activated. You might test with those two options deactivated and the sampling interval reduced to 2sec instead of the 1sec you run currently just to see if this makes any difference.

roasterdave, 2020-11-19:
> we want our event annotations to show during a roast
By design current roast annotations do not show while the roast is in progress.  Background profile annotations do display for reference during a roast (be sure to tick Roast>> Background> Annotations).  Once the roast is complete the background annotations will be hidden and the current roast annotations will be shown.  More information can be found [here](https://artisan-roasterscope.blogspot.com/2020/05/special-events-annotations.html).

shokunincoffee, 2020-11-20:
Thanks for the replies!
Firstly, in the Help > Errors, it shows the following a few hundred times (probably once every second):

500 12:37:45.560 Error: automaticsave() [Errno 24] Too many open files: 'C:/Users/jelle/OneDrive/Documents/Artisan\\4 El Sopapo 2020-11-18_1227.alog'@line 22605

499 12:37:44.391 S7 Communication Resumed

498 12:37:44.076 S7 Communication Error readInt: Communication error

Thanks for the tips about the Burner settings, I applied those right away and they fixed my issue. 

The Help > Serial log gives us the following:

210 09:34:14.937 S7 readFloat(5,4,6)

209 09:34:14.915 S7 readFloat(5,4,10)

208 09:34:14.587 S7 readInt(5,2,16)

207 09:34:14.587 S7 readInt(5,2,14)

206 09:34:14.587 S7 readInt(5,2,10)

205 09:34:14.587 S7 readInt(5,2,18)

204 09:34:14.587 S7 readInt(5,2,52)

203 09:34:14.587 S7 readInt(5,2,48)

Rebooting the pc didn't help; we tried roasting on three separate occasions in between which the pc was shut down. I'll try another roast next week without oversampling and see what happens.

Thanks to you, Dave, as well. Since we couldn't save profiles, we didn't get a chance to see the annotations in the background. I understand the reason for this setting and it will definitely work for us once we get things going without communication errors.

I'll give an update next week when we fire up the roaster again!

MAKOMO, 2020-11-22:
Thanks for submitting your logs, but they are not complete and seem to be recorded in different sessions based on those given timestamps thus they do not hint on exactly what goes wrong in the communication here.

For now I assume that the communication errors that re-occurs every second, once per sampling interval, is responsible for the "Too many open files" problem. It seems that per error a new connection is opened but the old one not closed. There is a known bug in the S7 communication lib that Artisan is using which can lead to this situation if the peer (in this case the roasting machine) closes the connection. I made some modifications now to potentially prevent this to happen, but I am not sure this really resolves the issue as I cannot test this for a lack of S7 hardware. The question remains what causes the communication errors in the first place. I guess it is either

a) the Giesen PLC receives too many requests and terminates some connections on internal timeout (are you running Cropster in parallel?)
b) PLC receives request for data it cannot answer and terminates the connection

to counterstrike a) I modified the Giesen WxA+ setup to disable oversampling (which is anyhow not productive in this setting) and modified the register map such that less requests are needed per sampling interval to retrieve all data.

to counterstrike b) I added better logging to the error and serial log

Could you please download and install the new continuous build that includes those changes from

https://github.com/artisan-roaster-scope/artisan/releases/continuous

Please run the menu Config >> Machines >> Giesen WxA+ setup again. Note: your own custom button definitions will be overwritten and you will have to enter again your machines IP address again.

Then please run a test session and attach here the complete error and serial logs. Note that you can upload text documents by a simple drag and drop here.

Thanks for your support,
Marko

shokunincoffee, 2020-11-25:
Hey Marko,
All the issues we had are fixed! We're doing our first Artisan production day now. Thank you so much for all the help.
I have three more questions:

- Is it possible to show the BT value in annotations with 1 decimal? Especially after First Crack this makes a huge difference

- Is it possible to program a button to turn on the Giesen's cooling tray? I'd love for it to turn on automatically when we drop a batch.

- Right now it's difficult to read the annotations at the end of the roast. The values for these gas settings are so close to each other that the comments overlap. The annotation overlap function doesn't seem to spread them out (whichever setting we try, they are either unreadable or some of them disappear). Is there any way to put them on top of each other?

[image]

MAKOMO, 2020-11-25:
> Hey Marko,
> All the issues we had are fixed! We're doing our first Artisan production day now. Thank you so much for all the help.

Good to hear. Thanks for testing!

> I have three more questions:
> 
> * Is it possible to show the BT value in annotations with 1 decimal? Especially after First Crack this makes a huge difference

Not sure about that "huge" knowing the average noise on such signals... The next version will show those with one decimal respecting the "Decimal Places" setting under menu Config >> Curves, UI tab. Thanks for the hint to make this more consistent.
 
> * Is it possible to program a button to turn on the Giesen's cooling tray? I'd love for it to turn on automatically when we drop a batch.

We would love to, just that we do not know which command would trigger this action. Giesen is not supporting us with this information and our own investigations did not lead to anything. If you have that missing information, just send it to us and we will add such a button.

> * Right now it's difficult to read the annotations at the end of the roast. The values for these gas settings are so close to each other that the comments overlap. The annotation overlap function [...]

The annotation overlap function tries to avoid vertical overlaps. Those horizontal overlaps are close to impossible to avoid without dropping all information, especially with long labels as you  use them. Note that you can click the individual event nodes on foreground and background profiles to have the event details listed in the message line. You can also consult the event table under menu Roast >> Properties or just choose a different way to render events, like the Combo mode. Your annotations only hold event value and BT. The BT can be read from the cursors x/y locator by moving the mouse in the events position and the Combo mode allows to render the event value directly at the event nodes. You might also try out the Step+ event mode.

> 
> [image]

closed 2020-11-25 by MAKOMO