artisan-roaster-scope/artisan #524: Change vertical axis label

aqdennis, 2020-11-25:
Hello,

We're very new users to Artisan and this week installed version 2,4,2 for our shiny new Giesen W6E.
The Giesen works with an RoR per 30 seconds so through a symbolic assignment I divide the Delta BT and Delta ET by two.  However, along the secondary vertical axis the label says "C/min"  while now that as become "C/30s".  Is there a way to change the axis label or if not, would it be a feature to implement in the future?

Thanks,
Dennis

MAKOMO, 2020-11-29:
As we believe that C/30s is not a proper physical unit we will refuse.

closed 2020-11-29 by MAKOMO

aqdennis, 2020-11-29:
How blunt. And you wonder why nobody donates.

Giesen works with RoR per 30s, you may like it or not, it's practice.