artisan-roaster-scope/artisan #526: Assign a background profile to a certain coffee

grebdnul, 2020-11-28:
## Expected Behavior
It would be really nice to be able to assign a background profile to a saved coffe type. So when you select File - New - and select a defined roast there, Artisan would automatically load your preferred background profile.
A "assign background profile" button in the Properties for the coffe-type would be nice, so you can select and change the background profile Artisan is going to load for that coffee.

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.4.2
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): 
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-11-28:
Them moment you create a new [Recent Roast](https://artisan-roasterscope.blogspot.com/2017/06/recent-roast-properties.html) entry, the current foreground profile gets linked to the new entry generated and listed also under menu New. once you select this entry that linked profile gets automatically loaded as background profile and the Roast Properties dialog gets filled with that data. If you select a Recent Roast entry from within the Roast Properties dialog, the profile linked to the selected entry gets loaded on leaving that dialog with OK. 

What is missing?

InvisibleOrdinary, 2020-11-28:
Hi Marko, 

I know this was brought up before, and closed without changing anything, but since this has come up again, and with using it now knowing how it works for the past several weeks, I see two huge benefits to amending the behavior:
1- easy reference to what profile is attached to the roast set up.
2- straight forward ability to update the profile attached to a coffee set up. As it's currently set up, you have to create&save a new 'coffee' with the profile you want loaded in order to update the roast profile. 
Tons of reasons why you'd want to do this, but obvious & important ones are: as a coffee ages, operator loaded the wrong profile the first time, change of heart on the right balance.
Also, from a UX / CX perspective, it's just not currently clear which profile is being used until you exit the properties dialogue. 
Thanks
Dave

grebdnul, 2020-11-29:
Ok Marko, I didn't know that it worked this way. My bad. I should have known. That is very good though! But it's not easy to understand, and not so straight forward maybe. You can't see which background curve a certain coffee has, unless you go new-and that coffe. However, you can just update the reference curve by loading it and then go to Properties for that coffe, load the preset you want to update background curve for, and press the +. That works for me!

closed 2020-11-29 by MAKOMO