artisan-roaster-scope/artisan #533: "Too many open files" after few runs

aqdennis, 2020-12-06:
## Expected Behavior

Save my profile

## Actual Behavior

Profile is not saved, message "Exception filesave(): [Errno: 24] Too many open files <filename>@line 26484

## Steps to Reproduce the Problem

  1.  Do a roast, save the profile afterwards
  2.  Reset, load last saved profile as background for new roast and do a new roast
  3.  Repeat 3 times, saving profile after roast no longer possible

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): a5caf1c8
  - Platform (Mac/Windows/Linux + OS version): Windows 10 Home
  - Connected devices or roasting machine: Giesen W6E (2020)
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/5648674/artisan-settings.aset.txt)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-12-06:
See Issue #523

MAKOMO, 2020-12-06:
Fixed in the v2.4.4 Release Candidate available here:
https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.4.4

aqdennis, 2020-12-06:
I am using 2.4.4.....downloaded yesterday.

MAKOMO, 2020-12-06:
Hm. This is bad news. Could you connect and run a few minutes and attach here your serial log (menu Help >> Serial, you need to tick the flag in that dialog to start logging)?

Thanks,
Marko

aqdennis, 2020-12-06:
Here you are, just logging warming up the machine.
[serial_log.txt](https://github.com/artisan-roaster-scope/artisan/files/5648817/serial_log.txt)

If you need something else, just let me know...

Dennis

aqdennis, 2020-12-06:
[serial_log_2.txt](https://github.com/artisan-roaster-scope/artisan/files/5648858/serial_log_2.txt)

A second one, with errors this time.

Dennis

MAKOMO, 2020-12-06:
Something does not fit together. Your settings attached above were written with Artisan v2.4.2 not v2.4.4. Your serial_logs indicate that Artisan is sending too many requests to the machine which Artisan v2.4.4 is not doing using your attached settings. Im puzzled. Could you please attach again the settings as exported from Artisan v2.4.4?

Anyhow, could you please deselect "Oversampling" (menu Config) and  in the S7 port settings (menu Config >> Port, 4th tab both flags, "optimize" and "fetch full blocks". This should resolve the issue for you.

aqdennis, 2020-12-06:
Ah, I yesterday updated from 2.4.2 to 2.4.4 but I just read the old
settings file, never saved the settings from within 2.4.4. That might
explain the mismatch in version number.
Will change the config as you mention...thanks!

Dennis.

aqdennis, 2020-12-07:
Deselecting Optimize (Fetch full blocks was already deselected) actually has the opposite effect: after a 10-12 minutes I get an S7 communiication error again. So right now I have Optimize back on but Oversampling off.
I have attached another serial log for your information.

Dennis
[serial_log_3.txt](https://github.com/artisan-roaster-scope/artisan/files/5652633/serial_log_3.txt)

MAKOMO, 2020-12-07:
Sorry I was in a hurry and did not write my sentence fully. You should turn off Oversampling and both flags, optimize and fetch full blocks on. That should resolve your issue.

MAKOMO, 2020-12-07:
If this does not help you need to switch to a 2sec sampling rate. Some Giesen PLCs are too slow and are not able to answer too many requests.

MAKOMO, 2020-12-07:
Could you send me also the message and error log of a period the issue happens (menu Help >> Errors, Messages). Thanks.

MAKOMO, 2020-12-08:
Dennis,

this one is mystic to me as I don't get a clear picture of what the machine returns (or not), what the real issue here is (I assume that the machine get's more requests than it can handle and that it terminates some of the connections, something that on the Artisan side might not be recognized because Artisan uses a separate C-lib for this communication) and I have no machine to test myself. I did apply some modification on the Siemens S7 that ensure that there is only one request per 0.03sec sent to the machine (kind of a random value, but quite moderate and corresponding to MODBUS standards over serial lines) and to better handle the case of terminated connections.

I would love if you could test the new build on your setup from here as this is blocking now the v2.4.4 final release:

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Thanks,
Marko

aqdennis, 2020-12-08:
Dear Marko,

I still have to roast tonight (Christmas madness here) so I will test it.
Any log file you would like me to generate an send back? Should I switch
on  oversampling again?

Cheers,
Dennis

MAKOMO, 2020-12-08:
Dannis,
excellent timing (of course bad for you with all that work;).
Please keep the oversampling off (this is not productive)( and best to keep both those flags (optimizer and full blocks) under Port ticked as this minimizes the number of requests send to the machine.

If things go north again, please send me all three logs the error, message and serial log (you need to tick that flag in the serial log before starting the session). 

Thanks for your support,
Marko

aqdennis, 2020-12-08:
Hi Marko,

First the good news .... I let it sample for the full warm up (35 minutes)
and did 2 roasts and no problems. I check the logs and  no errors....
I wrote a small program to check how fast I can sample the Giesen. I used
Snap7 library under FreePascal and used Cli_ReadMultiVars to sample all 8
values in 1 go. Taking 100 samples with 10 ms interval was no problem. I
had hoped of course to find that that was too fast....

The only strange thing now is that the first roast didn't give me
statistics or a summary afterwards. When I clicked on the area where it
should appear I got an error message. I have attached the message and the
profile...

The only difference I can think of is that for the first roast I filled in
some roast properties before the actual roast whereas for the 2nd roast I
didn't, I filled in the roast properties afterwards. Probably it has
nothing to do with it...

Cheers,
Dennis

MAKOMO, 2020-12-08:
> 
> 
> Hi Marko,
> 
> First the good news .... I let it sample for the full warm up (35 minutes)
> and did 2 roasts and no problems. I check the logs and no errors....

This is good news.

> I wrote a small program to check how fast I can sample the Giesen. I used
> Snap7 library under FreePascal and used Cli_ReadMultiVars to sample all 8
> values in 1 go. Taking 100 samples with 10 ms interval was no problem. I
> had hoped of course to find that that was too fast....

Thanks for testing this. So my limit of 30ms should be fine.

> 
> The only strange thing now is that the first roast didn't give me
> statistics or a summary afterwards.

Not sure what you mean by this.

> When I clicked on the area where it
> should appear I got an error message. I have attached the message and the
> profile...

Nothing attached. Could you send me both to check?

> 
> The only difference I can think of is that for the first roast I filled in
> some roast properties before the actual roast whereas for the 2nd roast I
> didn't, I filled in the roast properties afterwards. Probably it has
> nothing to do with it...

Could have to do something with it. Maybe something  you entered could not be rendered. Send me the screenshot and the profile and I will check and please keep an eye on the communication thing.

Thanks,
Marko

MAKOMO, 2020-12-08:
Note that you can attache files here (on the Github tracker) by drag-and-drop.

aqdennis, 2020-12-08:
Hi,

This is the profile. Just found out I didn't copy the screenshot to the
cloud so it's still on the laptop in the roastery.

But if you would read the profile it doesn't give the horizontal bar with
the 3 phases and the time/RoR/development per phase at the top of the graph
and neither the white summary area with the AUC on the right. When
right-clicking on the x-axis the axis label first gets the roast title
added, but when right-clicking again it does not give the advanced
information but an error message instead.
I can send you the screenshot tomorrow...

Cheers,
Dennis

MAKOMO, 2020-12-09:
Sorry nothing attached! You need to go to Github at https://github.com/artisan-roaster-scope/artisan/issues/533#issuecomment-741150466 and drag-and-drop it to the comment window at the very end of this page and press Comment. The Email that is send by the Tracker can be used to answer, but only for Text. Attachments are removed.

aqdennis, 2020-12-09:
Attached them to my earlier messages ....

MAKOMO, 2020-12-09:
Thanks a lot for this. Just fixed this. This is a long standing issue occurring only if the DROP event not being set. Thanks for all your support. I will close this one and release v2.4.4.

closed 2020-12-09 by MAKOMO