artisan-roaster-scope/artisan #535: Auto CHARGE not working

smaca69, 2020-12-08:
## Expected Behavior
When add green beans, it should detect the sudden drop in temperature and auto CHARGE should popup Roast Properties window.

## Actual Behavior
Auto CHARGE not triggered and Roast Properties window does not popup.  Have to manually hit the CHARGE button which skews the actual CHARGE temperature and actual start of roast.

## Steps to Reproduce the Problem

  1. Randomly happens
  2. 
  3. 

## Specifications

  - Artisan Version: 2.4.2
  - Artisan Build (number in brackets shown in the about box): 22ab78d9
  - Platform (Mac/Windows/Linux + OS version): Linux
  - Connected devices or roasting machine: Yoctopuce Thermocouple
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/5655982/artisan-settings.txt)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-12-08:
The autoCHARGE and autoDROP functions are based on heuristics that might fail on some (especially smaller) machines. We currently do not have plan to work on this as it works for most machines. In case you have an idea to make it work also on your machine without breaking it for any other machine, please raise a Pull Request. Thanks!

closed 2020-12-08 by MAKOMO

roasterdave, 2020-12-11:
Please check that you press the START button at least 5 samples before you expect the CHARGE event to be detected.  Artisan uses that data to detect CHARGE.  If you START too close to CHARGE it will not be properly detected.