artisan-roaster-scope/artisan #539: ALOG file includes incorrect Dry End information

DaveWCCanada, 2020-12-14:
I've written some code to read in the Artisan ALOG file and pull out the data that I am tracking such as Dry End time/temperature, FCs time/temperature and Drop time/temperature. These details are available towards the end of the file. But the Dry end data is forced to be the upper limit on the phase time & temperature for Drying. So while the curves will show 5:39 for time & 295F for temperature, and if I look at Roast - Properties - Events I'll see that event took place at that time/temp, within the ALOG file there is the data DRY_BT which will typically be (but not always) the 300F that I entered as the upper limit for Drying under Config - Phases - Drying. Is there some other way that I can pull this data out of the raw file? Why would you store the upper limit at all rather than the time/temp that I identified as the Dry End time/temp?

Thanks.

roasterdave, 2020-12-15:
Hi Dave, sorry I can not follow this.  Will you please provide **_all_** the information requested in the format shown when you clicked "New Issue".  Of particular importance are the steps to reproduce the problem, the information about your platform and Artisan versions, and a settings file and an alog file that demonstrate the problem.  Thank you.

DaveWCCanada, 2020-12-15:
## Expected Behavior  - I expected that when I looked within an alog file I'd see the actual time following DRY_time. Similarly, following DRY_BT I expected to see the temperature for Drying.

## Actual Behavior - In the attached file you'll see 'DRY_time': 369.9 which corresponds with 6:09. If instead of looking at the raw file, you load it in Artisan you'll see that the Dry time was 5:32. You'll see 'DRY_BT': 310.1 which roughly corresponds with the upper limit I set the drying phase to in Config - Phases rather than 298.1 which you'll see if you load the artisan file.

## Steps to Reproduce the Problem - virtually every file I look at has different information for dry time & temp in the raw data file compared to the same data in the artisan graph.

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:2.4.0 build 1 
  - Artisan Build (number in brackets shown in the about box): build 1 (164de804)
  - Platform (Mac/Windows/Linux + OS version): WIndows 10
  - Connected devices or roasting machine: Mill CIty Roaster 500g
[20-12-13_1534.txt](https://github.com/artisan-roaster-scope/artisan/files/5692523/20-12-13_1534.txt)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2020-12-15:
Thank you for the additional information.  It looks like you used a settings file supplied by Mill City.  While a configuration error due to those settings is likely the cause we can not support it.  Please refer [to this page](https://artisan-scope.org/machines/mcr/).  You should contact Mill City for support of these settings.

To use Artisan supplied settings do the following:
1. If you have made any custom configuration changes, save them to a file (menu Help>> Save Settings).
2. Upgrade to the [latest Artisan release](https://github.com/artisan-roaster-scope/artisan/releases/latest) which is v2.4.4.
3. Your settings should not have changed but you can always re-load the file you saved above  (menu Help>> Load Settings).
4. Load the applicable Artisan one click machine setup under the menu Config>> Machines.
5. Save the settings to a new file for safe keeping (menu Help>> Save Settings).

If the problem persists, please provide your settings file and a description for how to reproduce the error. In particular, how do you mark DRY END and do you move it during or after the roast?

To correct existing profiles that show the problem just open and then save them. 

We strongly suggest you use one of the File>> Export formats rather than try to access internal data which is subject to changes without notice.

closed 2020-12-15 by roasterdave