artisan-roaster-scope/artisan #540: Return installed version number from a bash script?

DavidFW1960, 2020-12-15:
## Expected Behavior
I would like to be able to get the version number of artisan installed via a bash script like I can do with any other linux program. This would enable me to get a notification when I start up the RPi running Artisan that there is a new version available to update to. Is there any command I can use to get the version from ssh?

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-01-12:
Thanks to FilePhil this is got resolved and will be part of the next version

DavidFW1960, 2021-01-12:
Thanks guys. Love your work..