artisan-roaster-scope/artisan #545: OT: Cheapest Roaster for USA

roaster-vic, 2021-01-03:
Hello community - does anyone know what is the cheapest roaster available in USA that is compatible with Artisan?

rich1n, 2021-01-03:
Hello @roaster-vic How many kgs? the [Aillio Bullet](https://aillio.com/?post_type=product) it's a great and cheap roaster; 1 kilo per batch.

PapasCup, 2021-01-07:
In the USA, and if you are looking at the Aillio Bullet, you'll want to visit the [Sweet Marias](https://www.sweetmarias.com/aillio-bullet-r1-roaster.html) website.  They are the US distributor for Aillio and they have lots of resources/videos on the product.  They are expecting a shipment of the Aillio Bullets toward the end of January.  They go quick (sometimes within minutes) every time Sweet Marias gets any in stock.  Get on their notification list and watch the site carefully as the expected time-frame nears.

I am a Bullet owner since March 2019, purchased from Sweet Marias and highly recommend it as a very good roaster.  I think there are other lesser cost, less capacity roasters than the Bullet where people have connected Artisan.  You can check out the [Home-Barista](https://www.home-barista.com/roasting/) discussion site to read what others are doing and engage them with your questions.  It's a good resource.

Regarding Artisan, it is the best choice for roasting software on the Bullet.  Aillio includes their own roasting software, but it is in the "junior league" from a capability view compared to Artisan.  I love Artisan and all the cool things that you can do with it.

MAKOMO, 2021-01-12:
Not an issue with the artisan software itself so I close it here and suggest to discuss this further on a public roaster forum.

closed 2021-01-12 by MAKOMO

PapasCup, 2021-01-12:
Hi Marko,

 

If I understand your reply correctly, you could not reproduce the problem?  

I installed Artisan ver 2.4.4 on (2) different PCs; a Notebook and a Desktop (with different displays) and saw the problem right away on both PCs.

I then installed ver 2.1.2 on both PCs and the problem is not there.  

I have been using Artisan for a couple years and upgraded the software several times.  

I’ve not seen the problem that I reported until I upgraded to 2.4.4 and it goes away if I remove it.  

 

Rick

MAKOMO, 2021-01-12:
Not sure about which issue you are talking. I answered and closed Issue #545 on the Artisan Issue Tracker.

PapasCup, 2021-01-12:
Oh, I was referring to the (only) issue I’ve reported recently which is #548 <https://github.com/artisan-roaster-scope/artisan/issues/548>  in the list.

 

In message #545, I didn’t realize that I would get an email message when you close someone else’s “issue” where I had only commented upon.

 

Thanks, I’ll wait to see what you find on the issue that I found and reported (#548).

 

Rick