artisan-roaster-scope/artisan #546: Linux mint -- can't add/save profiles

edgndg, 2021-01-04:
## Expected Behavior

My understanding from reading the documentation is that I should be able to save and reuse profiles from previous roasts by pressing the "+" button and/or selecting from the dropdown. Perhaps I am misunderstanding, though have read and re-read various documentation.

## Actual Behavior

My "+" button is grayed out.  When I save (press "ok"), I get the following message.  "Roast properties updated but profile not saved to disk"  I am assuming there may be some kind of file permissions issue (though I can save and open roasts), but having messed with various chown and chmod settings, can't figure it out...  Screenshots attached.

[image]
[image]

## Steps to Reproduce the Problem

  1. open roast properties
  2. change some values
  3. try to add/press save

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build (number in brackets shown in the about box): (34bc267)
  - Platform (Mac/Windows/Linux + OS version): Linux Mint 20
  - Connected devices or roasting machine: Phidget 1048
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/5765807/artisan-settings.txt)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

edgndg, 2021-01-04:
One other note -- I just went clicking through a variety of other items -- when I clicked on "Platform" under help menu it gave me this...  Not sure if related but adding in case diagnostically helpful

File "artisanlib/main.py", line 30208, in viewplatform
  File "artisanlib/platform.py", line 53, in __init__

--------------------------------------------------------------------------------
artisanlib/main.py::viewplatform:30208
	                self = <artisanlib.main.ApplicationWindow object at 0x7fb85e51b820>
	                   _ = False
artisanlib/platform.py::__init__:53
	                self = <artisanlib.platform.platformDlg object at 0x7fb8141989d0>
	              parent = <artisanlib.main.ApplicationWindow object at 0x7fb85e51b820>
	                  aw = <artisanlib.main.ApplicationWindow object at 0x7fb85e51b820>
	         platformdic = {'Architecture': "('64bit', 'ELF')", 'Machine': 'x86_64', 'Platform name': 'Linux-5.4.0-48-generic-x86_64-with-glibc2.4', 'Processor': 'x86_64', 'Python Build': "('default', 'Oct 15 2019 17:49:23')", 'Python Compiler': 'GCC 5.4.0 20160609', 'Python Branch': '', 'Python Implementation': 'CPython', 'Python Revision': '', 'Release': '5.4.0-48-generic', 'System': 'Linux', 'Version': '#52~18.04.1-Ubuntu SMP Thu Sep 10 12:50:22 UTC 2020', 'Python version': '3.8.0'}
	              system = Linux
	           __class__ = <class 'artisanlib.platform.platformDlg'>

MAKOMO, 2021-01-04:
The current Artisan version is v2.4.4 which fixes the platform bug listed above.  The RecentRoast mechanism is documented in a blog post: https://artisan-roasterscope.blogspot.com/2017/06/recent-roast-properties.html

You need to specify a charge wight and a roast title.

closed 2021-01-04 by MAKOMO