artisan-roaster-scope/artisan #548: Right side panel data digits not displaying corretcly for Aillio Bullet R1.

PapasCup, 2021-01-06:
## Expected Behavior:  Right-side data indicators for ET/IBTS, BT, BT ROR, Voltage, Exhaust ET, IBTS/BT should show the full digit/readout of the calculated data for these measurements.

## Actual Behavior:  The above data indicators (on the ride side of the screen) are missing the upper half of the digits.

## Steps to Reproduce the Problem

  1.   Installed version 2.4.4
  2.   Click on the "On" button which displays the data on the right-side panel to see the problem.
  3.   Note:  I've been using the same working setup/parameters from prior versions up through 2.1.2.  The missing data digits issue is not present with versions prior to 2.4.4.
  4.   I have attached the Zip file (below) that includes the .aset file and a .jpeg image to show the problem with the data digits.
[Artisan Bug report 01-06-21.zip](https://github.com/artisan-roaster-scope/artisan/files/5776805/Artisan.Bug.report.01-06-21.zip)

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): 1fb3d349
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine:  Aillio Bullet R1 with IBTS
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-01-22:
Sorry for this late reply. This will get difficult to analyse as non of the team members has access to an R1. Could you please explain a bit more in detail what you mean by those "upper half of the digits" which are missing. Maybe you can add a screenshot that illustrates this (just drag and drop one to the message text field for this issue on Github). Thanks!

PapasCup, 2021-01-22:
Hi Marko,

 

I did include a full screen-shot of the Artisan window within the Zip file that is attached to my issue submission titled “Artisan Bug report 01-06-21.zip <https://github.com/artisan-roaster-scope/artisan/files/5776805/Artisan.Bug.report.01-06-21.zip> ”.  There are (2) files within the Zip file, the .aset file of my settings and a .jpg file of the full Artisan screen-capture.  Here’s a snippet from that full screen capture for a quick look at what is happening to some of the digits being affected.  I have seen this problem on (2) different PCs (a Desktop PC and my Notebook PC that I use for roasting).  There are (6) of the data indicators that have the top half of the digits cut-off from view.  Let me know if I can provide anything else to help.

MAKOMO, 2021-01-24:
Dear PapasCup,

I am still somewhat confused. The screenshot tells me that Artisan is not able to talk to your R1 at all. Is the legacy USB driver correctly installed? There is no major difference between Artisan v2.4.2 and the current v.2.4.4 in regards to the R1 so I wonder if the Artisan v2.4.2 will work as expected on your setup. Could you please test by downgrading to v2.4.2?

If the USB legacy driver is not correctly installed, please re-install it using the following process with the Bullet plugged into your USB port:

* Download and run [zadig](https://zadig.akeo.ie/)
* You should see the `Aillio LTD - Bullet R 1 ROASTER FS` selected
* On the left side you will see your current driver
* On the right side pick `libusb-win32 (v1.2.6.0)`
* Install the driver

Please report!

PapasCup, 2021-01-24:
Hi Marko, 

I did as you asked and installed version 2.4.2 on both my Notebook and Desktop PCs to see if the problem with the digits was present.  Version 2.4.2 does not have the problem with the digits on either of the PCs.  So, from my perspective the issue seemed to be only associated with version 2.4.4.

**Problem Discovered:**  I decided to read through the "Detailed Release Notes" ("Changes" section) for version 2.4.4.  I wanted to see if I could find anything related to the digits problem I'm seeing on-screen.  Whala, I found a change listed for version 2.4.4 that states "**shows uu instead of -1 for error values in LCDs**".  That "uu" value makes the LCD value appear to be what a "0" would look like with the upper half of the digits cut-off.  So, that particular choice of LCD character representation is what caused me to think there was a display problem with some of my LCDs on-screen.

I connected my Notebook PC to the Aillio R1, applied power to the roaster, and launched Artisan 2.4.4.  Now, some of my LCDs are showing that "uu" symbol where they didn't with previous Artisan versions.  I have an "Omega HH506RA" Thermocouple Data Logger" configured as an "Extra Device" in Artisan that I use to measure Exhaust ET for my roasts.  That has worked fine (displays correctly in the LCD) in Artisan versions prior to 2.4.4.  With ver 2.4.4, all I get is that "uu" error in the Exhaust ET LCD.  So, version 2.44 did break reading temps from that Omega device.  I also noticed that the "BT ROR" LCD has the "uu" error at start-up.  I won't know if that LCD reading is broken too until I go through a roast where ROR data is active.

All-in all, I really don't like the "uu" change in 2.4.4.  I believe some other LCD digit value or just two zeros would be a better choice than "uu".  I can use version 2.4.2 which was before the "uu" error display change was implemented.

Additional information that you asked to be checked:
On my Notebook PC, I have had the correct USB Driver for the Aillio R1 for a couple of years, and roast using it with Artisan all the time.  The reason that you see the "Aillio R1: not found" in my screen capture is that I took the capture from my Desktop PC which is not connected to the Aillio R1.  I have Artisan installed on my Desktop PC to review and analyze my roasts that are done via my (smaller screen) Notebook PC.  The problem with the ver 2.4.4 missing upper-half of digits is exactly the same on my Notebook PC that does have proper USB connection to the Aillio R1.  I just took the screen-shot of the issue on my Desktop PC because I have better capture and processing tools on it.

I did use the "Zadig" USB utility and it shows that I have the "libusb-win32 (v1.2.6.0)" driver installed for the Aillio R1.  I have included a screen-capture (to this response) of the Zadig screen in case it helps.

[image]

MAKOMO, 2021-01-24:
Oh well. Only now I understand what you mean by "missing upper half". Yes, we replaced the misleading -1 by u to indicate "No Value Received". This is to end confusion as people thought that Artisan was receiving a -1. The choice for symbols on those LCDs are limited and after long discussions we settled to those u symbols that cannot be confused with proper numbers.

So I assume now that you can connect v2.4.4 to your R1 successfully and just got confused by the replacement of those -1 values by the u while Artisan is not receiving any data.

Regarding the Omega. I just checked back with mine and it connects without any issue to Artisan v2.4.4 as to all previous versions. Nothing changed on the Omega communication either. I assume that your Omega is also not reporting correctly on v2.4.2 for you. Please ensure that you have the proper serial port selected. Note that the Omega is really slow in delivering data (one response every 5sec only). I would suggest not to use the Omega in conjunction with the R1 as this would limit your data sampling rate also to the R1 to a max of 5sec.

I will close this issue now as the uu symbol thing was "resolved" and your R1 seems to communicate well. If there are any further issues, please open a separate issue for this. 

Thanks for your support, Marko

closed 2021-01-24 by MAKOMO

PapasCup, 2021-01-26:
Hi Marko,

 

After all the tests to upgrade and downgrade the recent versions of Artisan, the Port setting for my Omega HH506RA somehow got set to COM1 instead of COM3.  Thanks for your note on rechecking the COM Port setting and I reset it to COM3.  I did note that it only displays the Exhaust ET temp (from my HH506RA) if I have the Extra Device setting for that Datalogger set to “LCD2” in “Device Assignment”.  If I configure that Omega device to use LCD1, I get the error “uu” readout.  But configuring it to LCD2 reports the temp ok on-screen.  I thought that I would share that with you to see if you think it is a bug.

 

So, I’m going to try to use version 2.4.4 for an upcoming roast.  I noticed (on-screen) that the LCD for “BT RoR”  has the “uu” error when connected to the powered-on Bullet.  Since ROR would not be active just with the Bullet powered-on, I’m assuming that is the reason for the “uu” error?  I’ll know when I do a roast and ROR is active.  I’ll check to see that the “BT RoR” LCD starts reporting data when ROR is active on-screen.

 

Thanks,

 

Rick

MAKOMO, 2021-01-26:
> After all the tests to upgrade and downgrade the recent versions of Artisan, the Port setting for my Omega HH506RA somehow got set to COM1 instead of COM3.  Thanks for your note on rechecking the [...]

Well, the Omega features two inputs. Depending on into which port you connected your probe, it is available on channel 1 or 2 of that extra device. The other channel will report just this "uu" (no valid reading).

> So, I’m going to try to use version 2.4.4 for an upcoming roast.  I noticed (on-screen) that the LCD for “BT RoR”  has the “uu” error when connected to the powered-on Bullet.  Since ROR would not be [...]

The "uu" is not an error, it just indicates that Artisan has no valid reading (yet). While Artisan is not logging the BT signal, it cannot compute its RoR.

PapasCup, 2021-01-27:
Marko,

 

Thanks for the reply and helpful information.  

 

As you know, the HH506RA communicates via an antiquated communications technology using a Serial COM port to a PC.  I’m old enough to know what that is from my 43 years in the computing field.  😊  In my configuration, I had to connect the Omega unit via a “Serial to USB adapter” so that my Notebook PC could see the Omega data through a USB port.  So, not having a specific COM Port on my PC, I had no idea which one that Artisan is seeing via that adapter.  In my case, that had always been COM3 in Artisan and the device display setting was LCD1 in Artisan.  

 

The odd part is that I’ve been using Artisan for a couple of years that way by selecting “LCD 1” as the display for the Omega data.  After I started with the upgrades to 2.4.4 (from version 2.1.2), I’ve found that I had to switch that LCD display setting to LCD2.  Not a big deal since it works OK and is reading the Omega, just odd that it changed that way without my changing anything in my connection.

 

Anyway, I’m running 2.4.4 which was always my desire (to be on the latest version), and look forward to roasting with Artisan as always.  Thank-you for communicating to your customers and the timely replies of help.

 

Rick