artisan-roaster-scope/artisan #558: Problem with Artisan crashing on subsequesnt roasts

DavidFW1960, 2021-01-25:
## Expected Behavior

No crash on second roast
## Actual Behavior

Pretty sure Artisan is crashing. If I double click the artisan icon it tells me its running and wants to load a viewe=r if I click to close than it says I have to force quit.

## Steps to Reproduce the Problem

  1. after first roast complete
  2. I wait till temp ~70°C and set next roast (Heat/fan etc)
  3. start roast as per normal

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): (0)
  - Platform (Mac/Windows/Linux + OS version): RaspberryPi Debian Buster image
  - Connected devices or roasting machine: Hottop
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file. aset renames=d to.aset.txt so I could upload it. 
[alarms exported to text.txt](https://github.com/artisan-roaster-scope/artisan/files/5863850/alarms.exported.to.text.txt)
[alarms exported to text.txt](https://github.com/artisan-roaster-scope/artisan/files/5863852/alarms.exported.to.text.txt)

[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/5863897/artisan-settings.aset.txt)

attached
Please attach any relevant Artisan *.alog profiles. I don't use any. I just use alarms to alter the fan/hear at various temperatures and call script files which just publish mqtt messages to my broker (for notifications)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

****** I think this is related to new warn start in 2.4.4 as never any issue with any previous version and zero issue with for==irst roast.

DavidFW1960, 2021-01-25:
So I just shutdown Artissan after firs roast and restarted and it all seems to be working properly now. I ended up with 2 batches of beans ejected at 220 automatically.... crispy lol.

Anw=yway I'm sure it's 2.4.4 causing this and I'm =going to point my finger at the new alarm features causing it. So looking forward to what you can make of my files etc..

DavidFW1960, 2021-01-25:
actually just locked up!!!
[image]
[image]

DavidFW1960, 2021-01-25:
I rolled back to version 2.4.2 and just did 3 roasts back-to-back with no issue... perfect as usual prior to 2.4.4

MAKOMO, 2021-01-26:
David, just requesting a little more information:

1) After starting/rebooting your RPi, the first roast works fine as expected. All alarms trigger and the slider actions are reflected by the machine.

2) Only after you terminate Artisan after a first successful roast and restart it for the second roast, Artisan hangs on the first alarm-triggered slider action. Correct?

3) In case you do not terminate Artisan after the first successful roast and just start a second roast, does the roast runs successful or does it hang also on the alarm-triggered slider action?

I am asking those question as it could be that the root cause is the failure on fully terminate Artisan after your first roast leaving behind a hanging process. Note that Artisan starts a second process to organize the communication with the Hottop. If that one hangs, Artisan fails to communicate further with the machine which might lead to the hangs you observed. Just a theory....

DavidFW1960, 2021-01-26:
Honestly I can't be 100% sure. I did 8 roasts yesterday.
First one worked 100% fine.
Second roast locked up and I ended up (almost with ash) 100% sure about this point. I didn't terminate Artisan.
Third one... not sure TBH. I would have restarted Artisan for sure though because it hung/crashed. This is 100% your case 3 above and the second roast fails.
4th restarted Pi. I am not 100% sure this one didn't screwup as well.... maybe maybe not. I was watching like a hawk!!! I know one of these ones I pulled the USB plug and tried my best to eject and cool beans etc....
5th as above. Maybe 4th was OK and this one might have been when I took screenshots.
6/7/8 I had reverted to 2.4.2 and all was good...
If it hangs it's always at the point in the screenshot.

Probably not overly helpful right?

Normally I will do 4 roasts back-to-back and I never terminate Artisan between roasts. I just stop/reset the scope and then go again. (Press off button and then on and get the prompt to reset I say yes) Turn off fan, on heat and start. I get a message at 70 degrees to start the next roast before this. Pitch the beans at 100 etc.....

I hope this is some help.

DavidFW1960, 2021-01-26:
I actually thought the roaster was broken as this software is just so reliable and awesome!!

MAKOMO, 2021-01-27:
I checked all >400 changes to the main code file and tested your configuration with my Hottop using the relased v2.4.4 on my mac and cannot get this to hang in numerous runs. Test faster I deactivated  a number of irrelevant alarms and changed the conditions of those alarms with slider actions to trigger by time (early) and not by temperature. I also tested at different sampling rate as timing could be an issue too. No success. I am lost. No idea how to find and fix this one:(

Could it be related to the RPi vs mac?
Could you try to come up with a simpler alarm setup that can demonstrate this hang without having to burn through beans, like being able to run it without beans and at lower temperature to have faster turn around times?

[image]

MAKOMO, 2021-04-26:
Stall

closed 2021-04-26 by MAKOMO