artisan-roaster-scope/artisan #560: Feature Request: Designer

mtwenzel, 2021-01-27:
At the moment, the Designer approach is to create a BT curve and derive the DeltaBT == RoR from it. 

From my understanding, the aim in this is to shape the BT development such that it yields a desirable shape of the RoR curve, while respecting certain boundary conditions, like time of TP, or temperature after a certain time etc. 

The RoR curve is mathematically nothing but the first derivative of the BT curve, which means the BT curve is the integral of the RoR.

This again implies, with a little math (SciPy!) one could design the RoR curve directly, even simply by drawing it in the desired shape, and integrate the BT curve from this numerically. Introducing the constraints should then largely come down to a x/y shifting of the result.

Anyone interested in implementing this? I'd try my hand at this, but would need some intro where to do such calculations, as a quick look into the designer.py didn't help me, and also I don't really want to go into the UI development.

roasterdave, 2021-01-27:
Have you looked at this blog post?  [How to Create the Perfect Background Curve](https://artisan-roasterscope.blogspot.com/2020/06/how-to-create-perfect-background-curve.html)  
Maybe this method will allow you to create a BT curve that meets your needs.

mtwenzel, 2021-01-28:
Thanks Dave, that's a nice hint.
Different from what I'm suggesting however, it also shows a way to deal
with the "backward" design process, be it in a much more sophisticated way.
I guess I'll try to sketch a prototype of my thought as soon as I can get
to it.
In fact, I do get to a curve that is OK with the Designer as it is today,
but as a Newbie I found the process quite unintuitive and cumbersome. And
as I'm playful, I thought one could do it differently perhaps. So this
entry is neither urgent nor anywhere close to a bug, just a "call to
action" if someone would find it interesting.