artisan-roaster-scope/artisan #562: NodeJs interaction with Artisan

woudanator, 2021-01-29:
## Expected Behavior
I want to run commands to and from artisan to my nodejs application that controls my roaster 

Trying to stdout to push the ET and BT to artisan but it does not seem to pick it up

## Actual Behavior
The program runs according to artisan but it does not pick up or display the values 

## Steps to Reproduce the Problem

  1. go to device and program 
  2. insert "C:\Program Files\nodejs" D:/development/Roaster/artisanNode/aOut.js (in my case)
  3. and start the roast 
  
  here is my code I am using the stdout process instead of console log to make sure it simulates standard print 

```
  const process = require("process");

  const sendTemp = (eTemp,bTemp) => {
      process.stdout.write(`${eTemp},${bTemp}`)
  }

   sendTemp(13,12);
   
   Error in artisan logs is:
    
   56 10:46:36.149 callprogram() received: @line 714
```

your help with this will be much appreciated 

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): 1fb3d349
  - Platform (Mac/Windows/Linux + OS version): Windows 10 Pro 1909
  - Connected devices or roasting machine: Self Build roasters 

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/5892708/artisan-settings.zip)

roasterdave, 2021-01-30:
The error shows that nothing was received.  Most likely your program was not actually started.   Try adding quotes around your target program and change to all forward slashes as shown below. An example can be seen in the Help for Device External Programs.

`"C:/Program Files/nodejs" "D:/development/Roaster/artisanNode/aOut.js"`

Please confirm this works.  I also recommend you update your version of Artisan with the [official v2.4.4 release](https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.4.4)

woudanator, 2021-01-31:
Hi Dave 

That worked perfectly thanks so much for the assistance getting readings perfectly now

closed 2021-01-31 by woudanator