artisan-roaster-scope/artisan #567: Matching roast profile with tasting profile

eaguggi, 2021-02-01:
## 
I would like to have a sensory card database to transcribe the tasting observations and combine it with the roasting profile

## Actual Behavior: not allowed

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

FilePhil, 2021-02-09:
With Artisan you can create cupping profiles and add notes to the profil.
By using the "Report Ranking"-Feature you are also able to compare multiple roasting profiles.

Please Checkout the Links below.
https://artisan-scope.org/docs/cup-profile/
https://artisan-scope.org/docs/wheel-graph/
https://artisan-scope.org/docs/ranking-reports/

MAKOMO, 2023-11-20:
Seems to be already covered

closed 2023-11-20 by MAKOMO