artisan-roaster-scope/artisan #568: Artisan 2.4.4 is giving a lot of errors in windows 10.

fmpc, 2021-02-04:
### ****

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-02-04:
If you do not provide a minimal amount of information on your setup, what you are doing and what exactly happens we cannot track this down and will close this.

fmpc, 2021-02-04:
The errors are Windows blocks the hartisan, and when I'm toasting the I can't access the 500 error numbers functions. windows 10 carmomaq caloratto roaster.

MAKOMO, 2021-02-04:
Sounds to me like a configuration issue on your Windows system. Which version of Windows are you using? Maybe Artisan has not all rights. Did you try to run Artisan under an Admin account? Did you try to reboot your PC and test again?

fmpc, 2021-02-04:
I'm using Windows 10 Home Single Language, this as adiministrador i've done everything do not know but what to do.

roasterdave, 2021-02-04:
There is just not enough information provided to know what is happening.  A 500 error is typically a network server problem,  That is outside of Artisan.  Please post your Artisan settings file as requested above in the issue form.  And please post one of the Windows error messages so we see can see what Windows is reporting.

fmpc, 2021-02-04:
the error is this 
[image]

roasterdave, 2021-02-04:
This is an Artisan error that indicates a problem with your settings.  Does this error occur when you first start Artisan, or when you load a profile alog file, or when you load a settings aset file, or at some other time?  Again, please post your settings file as requested.  If you are loading a *.aset file please upload that file.  If you are not loading a settings file use these instructions from above to create a settings file:
```
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

Note that you need either add a .txt extension or zip the files before uploading. 
Otherwise you will receive a "Not a supported file type" error on uploading.
```
Do you see any other errors from Artisan?

You previously reported Windows is blocking Artisan.  Do you see errors coming from Windows, or is this the only error message you see?

fmpc, 2021-02-04:
It's these settings [configurações-artisan 2.4.4.zip](https://github.com/artisan-roaster-scope/artisan/files/5929120/configuracoes-artisan.2.4.4.zip)
[image]

roasterdave, 2021-02-04:
Please write answers to the questions in my previous post.  I need that information.

The Windows screenshot is information about the signing certificate for Artisan.  It is an information message and is not an error.

fmpc, 2021-02-04:
They're you
[configuracoes-artisan.2.4.4.zip](https://github.com/artisan-roaster-scope/artisan/files/5929191/configuracoes-artisan.2.4.4.zip)
](url)

roasterdave, 2021-02-05:
From my post above:
1. Does this error occur when you first start Artisan, or when you load a profile alog file, or when you load a settings aset file, or at some other time?
2. Do you see any other errors from Artisan?

Thank you for the aset file.

fmpc, 2021-02-05:
Every time I open the artisan it comes back in another configuration, in the ora of the roast I can not open any files and the buttons disappear.

fmpc, 2021-02-05:
today I torrei a lot of coffee, at the end of the day I looked at the errors gave 500 errors.

roasterdave, 2021-02-05:
Your settings file has some corrupt settings information.  This causes the settingsLoad() error and other issues.  Also, there is an error in one of your extra device equations.  This problem generates an error on every sample.  This is the reason for the full error log, which holds a maximum of 500 errors.

I repaired the aset file to fix these two problems.  Load this settings file (menu Help>> Load Settings), then quit Artisan.   Artisan is now ready to start and use.

[Repaired-configuraçSes-artisan 2.4.4.zip](https://github.com/artisan-roaster-scope/artisan/files/5933436/Repaired-configuracSes-artisan.2.4.4.zip)

fmpc, 2021-02-05:
Thank you very much for the help, now just appeared these errors.

version = 2.4.4

8 15:18:13.248 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

7 15:00:35.789 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

6 14:59:28.726 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

5 14:44:35.383 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

4 14:19:24.882 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

3 14:19:17.079 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

2 13:58:19.570 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

1 13:42:43.029 Exceção: OffMonitor() Invalid RGBA argument: ''@line 9312

roasterdave, 2021-02-05:
Do you see this error after loading the settings file I provided?

fmpc, 2021-02-05:
Yes

roasterdave, 2021-02-05:
In that case, you will need to do a factory reset before loading the settings file.

1. Use menu Help>> Factory Reset.  Click the reset button and Artisan will quit.
2. Start Artisan
3. Use menu Help>> Load Settings and load the settings file that I provided.

MAKOMO, 2021-02-08:
Any news on this?

fmpc, 2021-02-08:
I made the changes and it was perfect thank you very much.

fmpc, 2021-02-08:
I made the changes and it was very good did not appear but no mistake.

roasterdave, 2021-02-08:
This appears to be resolved.

closed 2021-02-08 by roasterdave

fmpc, 2021-02-09:
Yes it is solved was wondering if there is way to increase the numbers of buttons my artisan only goes up to 30.

roasterdave, 2021-02-09:
30 is the maximum number of buttons in each row.  Up to 4 rows of buttons can be displayed for a maximum of 120 buttons.

fmpc, 2021-02-10:
I'm with these mistakes I can't toast coffee.  3 17:25:07.057 Error: File "artisanlib\main.py", line 33147, in background File "artisanlib\background.py", line 143, in __init__ File "artisanlib\background.py", line 566, in createEventTable -------------------------------------------------------------------------------- artisanlib\main.py::background:33147 self = _ = False artisanlib\background.py::__init__:143 self = parent = aw = activeTab = 0 settings = backgroundDeltaETflagLabel = backgroundDeltaBTflagLabel = loadButton = delButton = alignButton = alignnames = ['CARREGAR', 'SECA', 'PEi', 'PEf', 'SEi', 'SEf', 'RETIRAR', 'ALL'] curvenames = ['', 'B3: Difer/Massa', 'B4: Extra 2', 'B5: Difer/Ar', 'B6: Extra 2', 'B7: Pre Crack', 'B8: Pos Crack', 'B9: DTR/Desenvolv', 'B10: Extra 2'] i = 3 __class__ = artisanlib\background.py::createEventTable:566 self = ndata = 17 start = 0 i = 0 

2 17:24:56.341 Error: File "artisanlib\main.py", line 33147, in background File "artisanlib\background.py", line 143, in __init__ File "artisanlib\background.py", line 566, in createEventTable -------------------------------------------------------------------------------- artisanlib\main.py::background:33147 self = _ = False artisanlib\background.py::__init__:143 self = parent = aw = activeTab = 0 settings = backgroundDeltaETflagLabel = backgroundDeltaBTflagLabel = loadButton = delButton = alignButton = alignnames = ['CARREGAR', 'SECA', 'PEi', 'PEf', 'SEi', 'SEf', 'RETIRAR', 'ALL'] curvenames = ['', 'B3: Difer/Massa', 'B4: Extra 2', 'B5: Difer/Ar', 'B6: Extra 2', 'B7: Pre Crack', 'B8: Pos Crack', 'B9: DTR/Desenvolv', 'B10: Extra 2'] i = 3 __class__ = artisanlib\background.py::createEventTable:566 self = ndata = 17 start = 0 i = 0 

1 17:24:13.675 Exceção: redraw() list index out of range@line 6818

fmpc, 2021-02-10:
Com esses erros não consigo construir perfil de torra.

roasterdave, 2021-02-10:
There is a problem with your background profile file.  Please post the background profile file (.alog) and the settings file (.aset).  

Does it work properly when there is no background profile?

fmpc, 2021-02-11:
Yes it works, it just doesn't work with the profile made in the builder.   [configurações-artisan 2.4.4.zip](https://github.com/artisan-roaster-scope/artisan/files/5967467/configuracoes-artisan.2.4.4.zip)
[Visor de Torra_21-01-26_0926 feira corrigido.zip](https://github.com/artisan-roaster-scope/artisan/files/5967468/Visor.de.Torra_21-01-26_0926.feira.corrigido.zip)

roasterdave, 2021-02-11:
The profile (.alog) is very broken.  It can not be fixed. 
It may be because your settings file (.aset) still has an error in the extra device definitions.  Please use the repaired aset file I uploaded previously.
Then, you will need to make another profile in the Designer.

For support using Artisan please use the Artisan mailing list (https://lists.einfachkaffee.de/postorius/lists/artisan-user.lists.einfachkaffee.de/) or post in a public forum such as https://www.home-barista.com/forums/