artisan-roaster-scope/artisan #569: I'm Brazilian as I do to acquire artisan pro.

fmpc, 2021-02-04:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-02-04:
If you do not provide a minimal amount of information on your setup, what you are doing and what exactly happens we cannot track this down and will close this.

MAKOMO, 2021-02-04:
Upps. Sorry. Please ignore my previous comment. I typed into the wrong window.

MAKOMO, 2021-02-04:
You just make an account on the platform at https://artisan.plus/en/ by clicking "CADASTRO" (REGISTER) and buy the "free" 3-month trial period from our shop:  https://shop.artisan.plus/en/professional-roasters/artisan.plus-professional-3-months-free

closed 2021-02-05 by MAKOMO