artisan-roaster-scope/artisan #572: Event Types couldn't be changed/saved

sc818, 2021-02-08:
## Expected Behavior
Event Types couldn't be changed or saved

## Actual Behavior
Event Types couldn't be changed/saved

## Steps to Reproduce the Problem

Changing Event Types: Air to Damper, Burner to Gas, then save the program, Restart it, but it still showing Air and Burner, can not be changed or saved.

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): 1fb3d349
  - Platform (Mac/Windows/Linux + OS version): Windows
  - Connected devices or roasting machine: 
  
Please attach your current Artisan settings file (as exported via menu Help >> Save 
[artisan-settings18.zip](https://github.com/artisan-roaster-scope/artisan/files/5947176/artisan-settings18.zip)

Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-02-15:
Works for me using your settings. Could you give us detailed instructions, maybe with screenshots, demonstrating how it fails? Note: your button labels will not change, they are text only and you have to change them yourself in the button settings.

MAKOMO, 2021-02-22:
Stall.

bkmr2ty, 2024-01-11:
I'm having this same issue, should I start a new issue, or can I add the info to this one? Thanks

MAKOMO, 2024-01-11:
Let's re-use this issue. I reopened it. Please attach your settings (menu Help >> Save Settings into an `artisan-settings.aset` file, rename it to `artisan-settings.txt `and drag-and-drop it here on the GitHub Web page to a comment box; reply by email does not work!).

Which version of Artisan are you using and which steps do we need to take to reproduce what you experience?

Thanks for your support!

bkmr2ty, 2024-01-11:
Ok, below is my version.

[image]

Here is my artisan settings file

[artisan-settings_v1- Copy.txt](https://github.com/artisan-roaster-scope/artisan/files/13905589/artisan-settings_v1-.Copy.txt)

My steps are:
I change settings on Config:Events I rename the Event types to the following:
[image]

I save my settings. I double check after saving and the settings are there.
When I close Artisan and re-open, the Event Types are reset to the default, my settings are gone:

[image]

I was able to get the saved Event Type settings to load by closing Artisan, then double clicking my settings file to start Artisan

I'm assuming when I open Artisan, that it will load with the last saved settings? Or do I have to load settings each time after opening Artisan?

Thanks for looking

roasterdave, 2024-01-11:
@bkmr2ty Are you on Windows?  How exactly do you quit Artisan (File>> Quit, or Click on the window X, or Alt-F4, or other)?

MAKOMO, 2024-01-11:
Thanks for your detailed report! I could reproduce the issue and fix it. 

While the issue is the same to the original reported one, your issue is caused by a bug in a change in settings handling introduced in the current version.

You can download the new [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) which does contain the fix.

closed 2024-01-11 by MAKOMO

bkmr2ty, 2024-01-11:
Great, Thanks!