artisan-roaster-scope/artisan #577: hartisan is giving error to open the run profile in the constructor.

fmpc, 2021-02-15:
[Visor de Torra_21-01-26_0926 feira corrigido.zip](https://github.com/artisan-roaster-scope/artisan/files/5982950/Visor.de.Torra_21-01-26_0926.feira.corrigido.zip)
[Repaired-configuracSes-artisan.2.4.4.zip](https://github.com/artisan-roaster-scope/artisan/files/5982955/Repaired-configuracSes-artisan.2.4.4.zip)

## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2021-02-15:
Sorry, I do not understand what problem you are reporting.  It will help to provide a step-by-step procedure to duplicate the problem.  

These files are the same files from [Issue 568](https://github.com/artisan-roaster-scope/artisan/issues/568)  

- 'Visor de Torra_21-01-26_0926 feira corrigido.alog' is badly broken and cannot be repaired.  It was created with a bad settings file.  You did not describe how this file was created.  
- 'Repaired-configuracSes-artisan.2.4.4.aset' is the repaired settings file I provided to you.

fmpc, 2021-02-24:
Sorry for the files.
I assemble the profile in the design, and save the file but when opening the profile background it doesn't open and these errors appear.

version = 2.4.4

2 20:57:42.853 Error: File "artisanlib\background.py", line 536, in load File "artisanlib\background.py", line 566, in createEventTable -------------------------------------------------------------------------------- artisanlib\background.py::load:536 self = _ = False curvenames = ['', 'B3: Difer/Massa', 'B4: Extra 2', 'B5: Difer/Ar', 'B6: Extra 2', 'B7: Pre Crack', 'B8: Pos Crack', 'B9: DTR/Desenvolv', 'B10: Extra 2'] i = 3 artisanlib\background.py::createEventTable:566 self = ndata = 22 start = 3.7903683 i = 3 timez = fmtstr = %.1f etline = btline = description = etype = evalue = 

1 20:57:42.849 Exceção: redraw() list index out of range@line 6929

roasterdave, 2021-02-24:
I cannot reproduce this.

Does this happen every time you create a profile in the Designer?  Can you repeat it?

roasterdave, 2021-02-24:
Please tell me exactly what you do to save the file.  Please be very specific about each step.

fmpc, 2021-02-24:
Yes, it happens every time I create a profile.
1- open the profile on artisan.
That phrase appears; foud a different set of extra devices.
To fully load this profile the extra device configuration needs to be modified
2 To save I go to file and save.

roasterdave, 2021-02-24:
Utilize o designer para construir o profile e depois guarde o profile.  Anexe o ficheiro criado pelo Artisan ao problema criado no Github. Capture um vídeo que demonstre o problema em questão. Precisamos de ver todos os passos desde que inicia o Artisan até o problema acontecer, incluindo o designer.

fmpc, 2021-03-01:
I'm sending the videos of artisan's design errors

https://user-images.githubusercontent.com/78544613/109511644-03061700-7a82-11eb-8f33-0ab401e49f46.mp4

https://user-images.githubusercontent.com/78544613/109511801-2fba2e80-7a82-11eb-9fd5-b2c061fd5747.mp4

https://user-images.githubusercontent.com/78544613/109511993-6728db00-7a82-11eb-9d6f-e9417c44ecdf.mp4

roasterdave, 2021-03-01:
Thank you for the videos.  I now see what is happening.  Until it is fixed please use this workaround.

Click Discard.  Do not click Cancel here.

[image]

roasterdave, 2021-03-02:
Thank you for reporting this problem.  There is a new continuous build that corrects the issue.  https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Please be aware that when using either v2.4.4 or with the new continuous build, you do not want to click Cancel.  The correct behavior for Cancel prevents the Designer from starting and returns to the normal graph.  Instead, you want to click either Save or Discard.  This will allow the Designer to start normally.

closed 2021-03-02 by roasterdave