artisan-roaster-scope/artisan #579: Old Equip, new equip

OldMan4114, 2021-02-21:
## Expected Behavior
Graph for both thermocoupler would work.

## Actual Behavior
Only appeared on right side of display. Gave message of "unable to get id from hh506ra device".

## Steps to Reproduce the Problem

  1. Turn artisan 1.3 on. Set device to omega hh506ra.
  2. Start roast, charge beans.
  3. Temps showed 0 or -0. Didn't graph/log.

## Specifications

  - Artisan Version: 1.3
  - Artisan Build (number in brackets shown in the about box): ca3c5f16
  - Platform (Mac/Windows/Linux + OS version): Windows 7
  - Connected devices or roasting machine: Omega HH506RA.
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/6018325/artisan-settings.txt)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-02-21:
Artisan v1.3 outdated. Only the current Artisan version is supported. Please update to the current v2.4.4 of Artisan. This one is also known to work with the HH506RA.

closed 2021-02-21 by MAKOMO

OldMan4114, 2021-02-21:
I thought artisan 2.4.4 only works with windows 10.
I have 7...

MAKOMO, 2021-02-22:
Microsoft ended support for Windows 7 on January 14, 2020. Time to update your system. Artisan v2.4.4 runs under Windows 8 and newer. However, future versions of Artisan will require minimally Windows 10.