artisan-roaster-scope/artisan #580: Artisan 2.4.4 "automatically" starting another roast

rmongiovi, 2021-02-24:
For the last few roasts Artisan has started acting differently when I end a roast.  I press the off button and then the reset and on buttons seem to automatically press themselves.  The screen clears and the graph of my previous roast is gone and I'm back to having just the off and start buttons displayed.

I'd don't know what I might have done to cause this except maybe the latest windows 10 patches.  Any ideas what might cause it and how to prevent it?

roasterdave, 2021-02-24:
Sounds like 'Keep ON' is ticked in menu Config>> Sampling.  Try clearing that option.

rmongiovi, 2021-02-24:
Yes!  Absolutely.  Thanks.  Since it was with the "sampling" settings I presumed it kept the connection to my thermocouples on.  I thought that might help with keeping my sampling smooth.  I have occasional intervals where the Artisan screen just stops updating for 15 or 20 seconds.  Obviously not.

Was it supposed to be obvious that it kept Artisan "on"?  'cause it failed entirely.  Maybe it could be a little less vague, like "Keep Artisan On."  Or maybe it should be on another popup entirely?

closed 2021-02-24 by rmongiovi