artisan-roaster-scope/artisan #582: Giesen W15A Communication Error - Connection Refused

gregs-legs, 2021-03-01:
## Expected Behavior

Ethernet connection to Giesen W15A (2013) 

## Actual Behavior

Error Message: S7 Communication Error readInt(5,2,16,False): b' TCP: Connection Refused'@line 593

Notes:
1)  I can ping the roaster from the command line.
2)  Giesen's software 'PC Link' can connect the roaster and correctly records probes etc.

## Steps to Reproduce the Problem

  1. Set Laptop to static ip 192.1682.42 subnet mask 255.255.255.0
  2. Configure > Machine > Giesen > WxA and select ip 192.168.2.199
  3. Turn on Scope

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box):  1fb3d349
  - Platform (Mac/Windows/Linux + OS version):  Windows 10 Pro on laptop
  - Connected devices or roasting machine:  Giesen W15A circa 2013/14 without touchscreen.  CPU: Moeller CP8-NT

MAKOMO, 2021-03-02:
Giesen roasters with Moeller PLC are not supported. Only machines produced from 6/2014 with Siemens PLC talking the S7 protocol are supported. See https://artisan-scope.org/machines/giesen/

closed 2021-03-02 by MAKOMO

gregs-legs, 2021-03-02:
My machine was built in March 2014 :-(
Could I make a feature request to support the Moeller PLC ?
Greg

MAKOMO, 2021-03-02:
The protocol implemented by the Moeller PLC is unknown. You will have to implement that yourself. Once you are done with the implementation you can make a PullRequest and I will add it to the source of the Artisan code.