artisan-roaster-scope/artisan #584: ARM64

PhatRik, 2021-03-07:
Adding ARM64 Support

MAKOMO, 2021-03-07:
This might take some time until all underlying libs and tools are updated with ARM64 support...

the-wallaby, 2021-04-26:
> This might take some time until all underlying libs and tools are updated with ARM64 support...

> Adding ARM64 Support

I second this.

MAKOMO, 2021-04-26:
Note that current Artisan builds as well as Phidget driver run under Rosetta 2 without issue on M1 Macs.

Dockside-cr, 2021-04-26:
Would this ARM64 architecture be compatible with the Raspberry-Pi 64 bit OS

kevinsithk, 2022-10-23:
> Adding ARM64 Support

I also need arm64 support for running on arm64 chromebook.
Thanks!

MAKOMO, 2024-08-09:
Note that support for AppleSilicon has been added in v2.10.0. ARM support for Linux, beyond the already existing RPi builds, is provided by the new Flathub release in v3.0. ARM support for Windows is pending.