artisan-roaster-scope/artisan #586: Websocket: Add ambient values (temperature, humidity, pressure)

Tobias007, 2021-03-11:
## Expected Behavior
Would be great, if we could feed via WS as well ambient data (temperature, humidity, pressure).

## Actual Behavior
WS can handle from my understanding only BT and ET. Other values can be put into new labels, but they will not be transferred into the roast attributes.

## Steps to Reproduce the Problem

Enhancement request.

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): (97e6dec) 
  - Platform (Mac/Windows/Linux + OS version): Mac
  - Connected devices or roasting machine: Own developed roaster electronic.
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

BillTheGoat, 2025-03-31:
Checking the source code for the [`Roast Properties` dialog](https://github.com/artisan-roaster-scope/artisan/blob/b1c0246162ce48cf7b6a00eccfcc7e30ec55e528/src/artisanlib/roast_properties.py#L1197), It calls a function on [line 4055](https://github.com/artisan-roaster-scope/artisan/blob/b1c0246162ce48cf7b6a00eccfcc7e30ec55e528/src/artisanlib/roast_properties.py#L4055), which attempts to update temp, pressure, and humidity. However, The pressure and humidity don't appear to be fully implemented.

We need data selectors for the other two fields, and to process all three on click. Actually reading the data is easy in the `websockets` pane of the `Ports Configuration` dialog.

neuralldev, 2025-06-08:
in 3.1.5 version, the update based on defined temerature field works. I retrieve if from Modbus. there are pressure and humidify as well. we just need that there is 2 other field selections and that the update button to push the 3 values into the field. it also happen to populate automatically after the roast so there is probably 2 places to update the code in this new version.

[image]

MAKOMO, 2025-08-19:
Added. See menu Config >> Devices, Ambient tab. Change is commited and will be available with the next [Continuous Builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) and will be part of the next official release. Hope you find time to test this somewhat and report back.

closed 2025-08-19 by MAKOMO