artisan-roaster-scope/artisan #596: Slider goes to 0 after pressing button to decrease

stefunkk, 2021-04-01:
## Expected Behavior
Slider should just go -1 after pressing button 

## Actual Behavior
Slider drop to 0

## Steps to Reproduce the Problem

  1. Add modbus integration sliders (just simple write([1,4,{}]) )
  2. Add modbus button value -1, value write([1,4,{}]) +-Air
  3. Show sliders in non-roast mode View/Sliders
  4. Set air slider to some value, ie 50
  5. Click On
  6. Click button to lower the fan
  7. Air slider drops to 0

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): 1fb3d349 (please make it possible to select and copy this value :)
  - Platform (Mac/Windows/Linux + OS version): Windows
  - Connected devices or roasting machine: Modbus integration with ESP8266, but it works like that even when nothing is connected
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

https://drive.google.com/file/d/1_ULLRqNoVbXp4lMr1_4IyIQB2YJgy34s/view?usp=sharing - settings

MAKOMO, 2021-04-01:
Negative event values are not supported

closed 2021-04-01 by MAKOMO

stefunkk, 2021-04-01:
It does. When i set slider manually after pressing "On" it works as expected.

stefunkk, 2021-04-01:
It works the same when :

2. Add modbus button value 1, value write([1,4,{}]) +-Air

MAKOMO, 2021-04-01:
Relative event value changes are always relative to the last set value. The last set value is initialised to 0.

stefunkk, 2021-04-01:
But slider shows 49%, fan is working on 49%, and then you press increase fan button, and fan drops to 0. It's really dangerous with fluid beds.

stefunkk, 2021-04-01:
Is there any workaround to fix it with alarms? I nearly broke my heater today