artisan-roaster-scope/artisan #597: Slider goes to 0 after pressing button to increase

stefunkk, 2021-04-01:
Expected Behavior
Slider should just go +1 after pressing button

Actual Behavior
Slider drop to 0

Steps to Reproduce the Problem
Add modbus integration sliders (just simple write([1,4,{}]) )
Add modbus button value 1, value write([1,4,{}]) +-Air
Show sliders in non-roast mode View/Sliders
Set air slider to some value, ie 50
Click On
Click button to increase the fan speed
Air slider drops to 0
Specifications
Artisan Version: 2.4.4
Artisan Build (number in brackets shown in the about box): 1fb3d34 (please make it possible to select and copy this value :)
Platform (Mac/Windows/Linux + OS version): Windows
Connected devices or roasting machine: Modbus integration with ESP8266, but it works like that even when nothing is connected
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

Note that you need either add a .txt extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading.

https://drive.google.com/file/d/1_ULLRqNoVbXp4lMr1_4IyIQB2YJgy34s/view?usp=sharing - settings

MAKOMO, 2021-04-01:
Understood. Just modified this behavior to keep the previous set event value on pressing ON or RESET (Commit 985fbc879f56c583d3cb502a37af8e691530a284). A new beta build v2.4.5 should appear in a few minutes here: https://github.com/artisan-roaster-scope/artisan/releases/continuous

closed 2021-04-01 by MAKOMO

stefunkk, 2021-04-01:
Wow, that was quick, thank you. Is there any other way of donating? I have paypal account blocked.

MAKOMO, 2021-04-01:
Donations are only possible via PayPal. Sorry.