artisan-roaster-scope/artisan #598: Feature: Rename "start" and "on" buttons

garthwebb, 2021-04-04:
The buttons controlling machine monitoring and roast recording are confusing.

## Expected Behavior

A) I know which button to click when I want to start monitoring my roaster because the button is labeled "Monitor".  When I click it, it changes color to indicate it's active.
B) I know which button to click when I want to start recording a roast because the button is labeled "Record".  When I click it, it changes color to indicate it's active.  The "Monitor" button greys out to indicate this can't be turned off while recording is active.

## Actual Behavior

A) I never remember the difference between "on" and "start" so I mouse over each button to read each tool tip and then click "on".  When I click it, the text changes to "off" which I also have to mouse over to remember.
B) I never  remember the difference between "on" and "start" so I mouse over each button to read each tool tip and then click "start" (even after doing A above).  When I click it, "start" greys out and the "off" button which used to be "Monitoring off" suddenly becomes "recording off", furthering confusion.

## Steps to Reproduce the Problem

  1. Start up Artisan
  2. Start monitoring a machine
  3. Start recording a roast

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): 97e6dec
  - Platform (Mac/Windows/Linux + OS version): Mac OS, Catalina, 10.15.7
  - Connected devices or roasting machine: Hottop KN-8828B-2K+

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/6253816/artisan-settings.aset.zip)

MAKOMO, 2021-05-28:
Request comes 12 years to late. Rejected at this point. Try to remember...

closed 2021-05-28 by MAKOMO

garthwebb, 2021-05-29:
Your answer isn't very thoughtful given I spent a lot of time to document this issue and gave a suggested fix.  If you're saying that the UI reached perfection 12 years ago, and there is no possible improvement possible, this should be made clear in the bug report template. "No UI suggestions accepted".  Or maybe you should disable new issues altogether so nobody else wastes their time trying to give constructive feedback:

https://docs.github.com/en/github/administering-a-repository/managing-repository-settings/disabling-issues

MAKOMO, 2021-05-29:
My comment was not detailed enough. Suggestions to improve the UI are welcome. However, not all suggestions can and will be realised. The names of those fundamental default buttons have been well established in a large user community over a long period and sounds like to request a rename of "Windows" into "Doors". Renaming such fundamental structures at this point is critical and will cause lots of confusions besides that one needs to adjust 30+ translations accordingly and find corresponding terms in those languages too. This is not doable considering the very limited resources the project has, which are better spent to fix major bugs or add useful features. In case you insist on different button labels, just fork and change those to your likes in your copy. This is open source after all.