artisan-roaster-scope/artisan #602: Feature Request: Lock/hide settings

fcebreros, 2021-04-08:
I would be great to be able to lock or hide device, port, events and temperatures settings (and alarms perhaps) with an administrator password or with a hidden option somewhere .
This would avoid operators to mess up settings and break communication/functions by error.
Thanks
Frank C.

poundy, 2021-05-16:
Can you avoid this by implementing a saved settings file that is always the "master" ?  If an operator messes with things, they just fall back to the approved common settings file...

Robjones6146, 2023-09-30:
Great suggestion. I would say a lock/hide for the whole **Config** dropdown options (Machine, Device, Port..... all 16 )

poundy, 2023-09-30:
> Great suggestion. I would say a lock/hide for the whole **Config** dropdown options (Machine, Device, Port..... all 16 )

after more than two years, with only the original feature suggestion and now yours, it should be clear there's not much support for this so it'd be hard to justify addressing this unless someone contributed the code or their programming time... It might help if you helped identify why you think that it's important, what made you suggest this, as it's not clear from the (lack of) interest shown that it'd be worthwhile.

Robjones6146, 2023-10-01:
To be honest I didn't see the date. The first time I looked at the software, I was surprised to see that all the settings were exposed and so easy to change. I have 44 events and 38 alarms controlling 14 microcontrollers. One careless typo could cause havoc with the automation. I am not the end user. A preference menu in the future is not a bad idea.
[image]

I have made a few requests, some implemented, some rejected. I tend to do a thorough search for possible solutions and for ways that I might not have considered to achieve a result before posting a request.

Always foremost in my mind is the amazing work done and the effort that goes into Artisan. Artisan is the reason I designed and built a roaster. Bug fixes should always come first, good ideas that will solve problems for lots of users next, and then "Nice to have features" last.