artisan-roaster-scope/artisan #604: Modbus Ascii configuration

oscarquiroz94, 2021-04-23:
Dear.

I'm trying to configure Artisan to connect a roaster using Modbus over serial port (ascii).

But I can't read the temperature values properly.
The roaster uses the following configuration on the Modbus series:

**Baud rate:** 38400
**Function:** 3
**Storage format temperatures:** decimal with 2 digits (16 bits)
**Slave id:** 1
**Register1:** 100
**Register2:** 101

My configuration in Artisan is:
[image]

I'm doing a test with temperatures 180.74°C and 236.59°C for example, and I get the following:
[image]

I can see the temperatures values in the serial log (RX: 18074 and RX:23659) but in the LCD temperatures ET and BT are differents.

fcebreros, 2021-04-23:
HI,
I think you should have the
*Decode* option on *Int*
and the
*Divider* on *1/100*

Best
FC

oscarquiroz94, 2021-04-24:
Hi.

The only option in Decode is IntFloat, of course I'll put 1/100 in the divider. 

[image]

I´ll try it on Monday and I'll tell you about it.
Regards

MAKOMO, 2021-04-24:
On older version the empty Decode entry assumed regular integer register access. Select the first, empty entry. Better: update to the current version which is v2.4.4.

M

oscarquiroz94, 2021-04-26:
Hi.

I downloaded the new version of Artisan, this is the configuration that works for me:
[image]
[image]

Thank you.

MAKOMO, 2021-04-26:
Not an issue in the Artisan software. Invalid.

closed 2021-04-26 by MAKOMO