artisan-roaster-scope/artisan #608: Quantifiers changes not being logged to Events consistently

bravowren, 2021-05-07:
**Describe the bug**
When Artisan is configured to log Events using quantifiers, if one uses keyboard inputs on PC to trigger an Event,  corresponding Event quantifier inputs are not logged to events for a specific period of time (approx 45 seconds) after keyboard input, even though they are visible in LEDs and tracked by curves. Event logging from quantifiers resumes normally after this 45 second lag.

**To Reproduce**
1. Click "Start"
2. Change quantifier inputs - works as expected in triggering and logging Event.
3. Use keyboard shortcut or button or Slider to trigger Event
4. Further quantifier input is displayed on curve and LED, but not logged to Event - event logging resumes normally in about 45 seconds.
This behavior is 100% repeatable, and it has been reproduced in both 2.4.4 and 2.4.5(beta), using both the Mac build, and Win build.

**Expected behavior**
Event changes logged should always follow quantifier changes as configured in Events>Quantifiers.

**Screenshots**
1. Profile showing quantifier curves and corresponding logged Events.
2. Event Log.

**Setup (please complete the following information):**
 - Artisan Version 2.4.4 AND 2.4.5 (beta)
 - Artisan Build e4ae7bc(2.4.5) and 97e6dec (2.4.4)  
 - Version
 - Platform Mac 10.15.7 and Win10 (emulator on Mac)
 - Connected devices or roasting machine: Phidgets Hub, Phidgets OUT1100, 10K linear taper potentiometers, TMP1100 (x2)

**Additional context**
I have 2 potentiometers connected to Phidget Hub analog voltage inputs. These voltage inputs drive quantifiers, that in turn trigger Event changes. Potentiometers work as expected - until an event is triggered on the PC keyboard (by button, keyboard shortcut, or slider). Following such keyboard input, further quantifier inputs for that Event are not logged as Events for a period of approx. 45 seconds, after which quantifier inputs resume tracking to Events. 

If you look at the enclosed profile screenshot, you will see the quantifier curves tracks events as expected until 1:00 min, when there is a keyboard input for "Air 12" event/value. Two following quantifier "Air" inputs clearly shown on corresponding curve do not have corresponding Events logged. Event logging resumes tracking quantifier at 1:45 min. 

Same thing occurs with "Drum" Event logged at 2:57. Quantifier input Event logging does not resume until 3:42.

[image]
[test_artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/6439164/test_artisan-settings.aset.txt)
[testprofile.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/6439165/testprofile.alog.txt)
[image]

MAKOMO, 2021-05-07:
This is on purpose for the application this quantifier-based synchronization mechanism was developed. The intended use case is to connect a roasting machine that features its own controls. Control inputs on Artisan are send to the machine which changes its state. To avoid that the state changes on the machine side observed by the quantifier mechanism trigger an (unintended sequence of event changes on Artisan until the new state is stable (on some machines changing the burner level from 10% to 100% takes several seconds while the machine reports many intermediate states). This is a work around for machines that do not report the set value (SV) for actors independent of the process value (PV). I will add (another) flag that allows to deactivate this "delay".

MAKOMO, 2021-05-07:
Here is a new build with that SV flag in the Quantifier setup that you should tick to deactivate this "delay".

https://www.dropbox.com/s/ackj79ndya9n1vo/artisan-mac-2.4.5.dmg?dl=0

Please report if this works.

MAKOMO, 2021-05-28:
resolved

closed 2021-05-28 by MAKOMO

bravowren, 2021-05-28:
Marko, in looking back at my last comment, I'm sorry I did not make it clear that this works perfectly. Thanks again!