artisan-roaster-scope/artisan #609: Control Button Crash

hertizonical, 2021-05-08:
## Expected Behavior
When I click "Control" it used to open the PID controls. But...

## Actual Behavior
Now, the program just hangs. I can't click on anything. I have to close the program through the task manager.

## Steps to Reproduce the Problem

  1. I tried turning off my computer and unplugging all peripherals. On reboot. Same result
  2. Tried upgrading to 2.4.4. This also led to the same issue.
  3. I can operate everything but the PID. I tried clicking the control button before turning the machine "on" and same result.

## Specifications

  - Artisan Version: 2.4.4 and 2.4.0
  - Artisan Build (number in brackets shown in the about box): 1fb3d349
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Arduino TC4
  - 
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/6445182/artisan-settings.aset.txt)

version = 2.4.4 (1fb3d349)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.19041-SP0
Processor = Intel64 Family 6 Model 126 Stepping 5, GenuineIntel
Python Branch = tags/v3.8.0
Python Build = ('tags/v3.8.0:fa919fd', 'Oct 14 2019 19:37:50')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = fa919fd
Python version = 3.8.0
Release = 10
System = Windows
Version = 10.0.19041
Win32 = ('10', '10.0.19041', 'SP0', '')
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-05-08:
I cannot duplicate this one. If I load the attached settings into my Artisan v2.4.4 (v2.4.0 is no longer supported) and press Control, the dialog popups up and I can operate everything, change settings and close the dialog again without any issue. Do you observe the issue also if you disconnect the hardware and press Control directly after starting Artisan? If not, your issue might be a communication issue to the TC4. Note that the PID dialog sends changed settings to the connected TC4. If this communication fails the dialog might hang.

MAKOMO, 2021-06-03:
Stale

closed 2021-06-03 by MAKOMO

hertizonical, 2021-06-04:
Sorry for not getting back. These emails ended up in a spam of sorts.

I ended up resetting the program back to factory and reprogramming from scratch. That seemed to work. Something in the setting must have got corrupted somehow. It was very weird how it worked for one batch and then stopped on the second batch immediately following.

Either way, we are back up and running. Thank you for taking a look!

Closed #609<https://github.com/artisan-roaster-scope/artisan/issues/609>.