artisan-roaster-scope/artisan #616: Add Between Batch Protocol (BBP) in the Background Profile

Nicodmk, 2021-05-27:
Usually background profile can be load on start only, based on artisan blog's "Supporting Between Batch Protocols", now the all the profile have bbp record (because of the auto start/keep on mode). Can you make the background profile have the option to align since "start mode" on ?

fcebreros, 2021-05-27:
Hi,
not sure what you mean but on the right bottom corner you select how
the background profile gets aligned.
[image: image.png]

Nicodmk, 2021-05-30:
Sorry If I make you confused, It can be aligned from "charge to All", there is no option from "on" or "start". Thanks for responding

MAKOMO, 2021-06-03:
We change the alignment of a new recording to a loaded background to start at begin of background profile instead of its CHARGE mark in the next version. Thanks for the hint.

closed 2021-06-03 by MAKOMO