artisan-roaster-scope/artisan #617: Power setting reading and recording sporadically

AnonymousRoaster, 2021-05-27:
## Expected Behavior
power level displayed and recorded on graph 

## Actual Behavior
power level records on screen but does not plot on graph. Only registers for one or two roasts then does not register

## Steps to Reproduce the Problem

  1. start roast
  2. turn on gas - sometimes will register, often does not
  3. adjust gas(power) no change registers on screen (power value) or graph

## Specifications

  - Artisan Version: Artisan 2.4.2 build 3 (bcaf44d) 
  - Artisan Build (number in brackets shown in the about box):build 3 (bcaf44d) 
  - Platform (Mac/Windows/Linux + OS version):Mac (Macbook air) macOS Big Sur v11.2.3
  - Connected devices or roasting machine: Diedrich IR12 (6 sensor)

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles. 
[artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/6551802/artisan-settings.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

CarefreeBuzzBuzz, 2021-05-27:
This is unlikely an Artisan issue.  Sounds more like a device issue.  You didn't describe which of the six sensors aren't working and what devices they are and how they are configured.  Is this a standard Diedrich set up?  What did they advise you?  Do you have a configuration  file  from them or a prior owner that worked?

MAKOMO, 2021-06-10:
Stalled

closed 2021-06-10 by MAKOMO