artisan-roaster-scope/artisan #626: Artisan control of Phidget DCC10xx missing an essential command

bravowren, 2021-06-09:
**Describe the bug**

I understand that Artisan only supports basic functions on Phidget DC Motor Controllers (DCC10xx)- velocity, acceleration, & sleep. But motors drawing more than 2A can not be controlled from Artisan using these commands. This is because these Phidget controllers have a default setting that limits their current output, even though they are rated at much higher current.

The DCC1000 (max 25A) has a default current limit set to 2A as seen in the Phidgets Control Panel.
The DCC1003 (max 4A each motor) also has a default setting of 2A according to Phidget tech support, though the MacOS Phidgets Control Panel does not show this slider and the WIN Phidgets Control Panel shows it at 0A.

I was able to reliably start the drum motor on my Cormorant CR600 roaster using either DCC1000/1003 from the Phidgets Control Panel only if I increased the current limit to 2.5A or greater. At the default setting of 2A on the DCC1000, it would only start intermittently, though about 90% of the time it would not start and you could hear it laboring. This was without any beans in the drum, which would have increased the starting torque required and the subsequent current draw. The DCC1003 would not start up the motor at the default setting from the control panel.

**To Reproduce**

I was unable to start the same motor using the max "vel" command from Artisan, because I had no ability to increase the current limit on the DCC10xx. 

Even though the operating current of my motor is right below the limit threshold, startup torque on these gear motors requires higher draw for a few ms, which triggers the sensitive Phidget current limiter.

(BTW Artisan "vel" command worked perfectly on the same setup with with a lower load, as observed viewing the output PWM duty cycle). 

**Request**

Marko, is it possible to add the SetCurrentLimit API command? My understanding is that it only needs to be sent once at the beginning of a session. I'm happy to send you a controller for testing if you need it.

**Setup**
Phidgets DCC1000 (v211) 
Phidgets DCC1002 (v110) 
Cormorant CR600 roaster: 14V 1.8A drum motor (starting torque >2A)
 - Artisan Version v2.4.5 
 - Artisan Build 6c69312
MacOS 10.15.7
Win 10

MAKOMO, 2021-06-10:
I added the "limit" command

`limit(ch,v[,sn])` 

to set the current limit.

Could you please test if this works as expected? Here is a per-build:

https://www.dropbox.com/s/bdfmvc5ir74ymce/artisan-mac-2.4.5.dmg?dl=0

bravowren, 2021-06-10:
Just tried it. Unfortunately No Go.
Tried a couple of different ways:
1. slider IO command: "limit(0,4);vel(0,{})"
2. button commands: "limit(0,4) proceeded by vel(0,1)

To confirm, correct output is showing at output with motor unplugged.

Hardware setup is good - ran the motor fine on the exact same setup from the Phidget Control Panel (had to switch to Windows). Had to bump the Current Limit to 3V to get it started.

I received this python script to test the motor from Phidget's tech support a couple of days ago, to run on my Mac, as the Phidgets Mac Control Panel does not have the Limit control. Not sure if it is of any help to you.

from Phidget22.Phidget import *
from Phidget22.Devices.DCMotor import *
import time

def main():
	dcMotor0 = DCMotor()

	dcMotor0.openWaitForAttachment(5000)
        dcMotor0.setCurrentLimit(4)  
	dcMotor0.setTargetVelocity(1)

	try:
		input("Press Enter to Stop\n")
	except (Exception, KeyboardInterrupt):
		pass

	dcMotor0.close()

main()

MAKOMO, 2021-06-11:
Found a typo in the API call I added: ".SetCurrentLimit" => ".setCurrentLimit". Hope this was the blocker. Still untested as I lack the hardware to do so. Sorry. Could you give this another test?

https://www.dropbox.com/s/bdfmvc5ir74ymce/artisan-mac-2.4.5.dmg?dl=0

bravowren, 2021-06-11:
It works perfectly now!

FYI, if I send the IO "limit" command once at the beginning of a session, the limit does not change throughout the session, even when moving the velocity to 0 and back up. 

The slider IO command "limit(0,4);vel(0,{})" also works fine, and I may just use this, unless you think the redundant limit commands will slow things down.

Thanks again Marko for your incredible work!

CarefreeBuzzBuzz, 2021-06-11:
Please make a contribution for this custom service if you haven’t already.  That’s the only way innovation continues.  

https://artisan-scope.org/donate/
----------------------------------------
Michael Herbert
***@***.***

closed 2021-06-11 by MAKOMO

bravowren, 2021-06-11:
@CarefreeBuzzBuzz  absolutely agree, and have done so. 

@MAKOMO Interestingly, even though motor max running current draw was measured at 1.8A, the DCC1003 output at 100% duty cycle when the limit was set to 2.5A was not clean (viewed on a scope), and the running motor sounded different when the limit was set at 2.5A vs. 3A or higher. The DCC1000 had a cleaner output at 2.5A limit. You may want to include advice in the next release documentation, to set the limit at a minimum of 50% above motor nominal rating. Thanks again!

MAKOMO, 2021-06-12:
Thanks fo your donation and the hint on setting the limit vs nominal motor rating. This is somewhat external to the Artisan software and should be documented by Phidgets to my opinion. Artisan is just forwarding the users values via the API to the module. How the module handles those values is kind of transparent to the software and thus out of its scope.

ronncat, 2021-06-13:
So I just joined github to comment on this. I have been working on automating my Cormorant and had the same exact problem trying to get the Phidgets DCC to work with my drum motor. I started a thread on Home Barista asking for help. Then I stumbled across this. I have very little experience programming ( have done a couple of Arduino projects) so haven't a clue how all this github works. Is this fix something that you will incorporate into Artisan in the near future?? Or is it possible to easily add and test this result into my existing setup?? I appreciate your feedback and am more than willing to contribute to the cause again. Thank you....  Ron

roundfile, 2021-06-13:
@ronncat The changes discussed in this topic will be a part of the next Artisan release (no date set).  What platform are you using?  If it is the Mac there is a link to a test version above that you can download.  A test version for another supported platform can be provided, let us know.

Contributions allow Artisan to continue innovating and developing.  Thanks for your support.

MAKOMO, 2021-06-13:
The above link  for the macOS version including this fix is still valid (for some days). The Windows build is here:

https://www.dropbox.com/s/2yvmd9eczm13zxo/artisan-win-2.4.5.zip?dl=0

ronncat, 2021-06-13:
Marko,

I have just installed the Windows build on my Windows 10 laptop and plugged it in.  SUCCESS😁😁😁.  Thank you. I have also made a donation to the cause. Please keep up the great work and Thank You so much for your incredibly prompt response and your help. You are the man!!!

Ron

bravowren, 2021-06-13:
@ronncat , I will be posting some detailed info on the Cormorant Facebook group in the next couple of days re Artisan control of Cormorant using Phidgets. You can PM me there. Gavin Martin

ronncat, 2021-06-13:
Thanks Gavin. Got it working… I am only able to see the Facebook group via my wife’s account as I’m not on Facebook. She is a member though as I begged her to join