artisan-roaster-scope/artisan #629: Opening the program on my IPad

MakenuChocolate, 2021-06-17:
Hi, I am having an issue on installing artisan onto my 8th generation iPad. when I download artisan-mac-2.4.4 2 nothing happens. Page opens up but there are no icons to open. Any help would greatly be appreciated on how to open the program on my IPad

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: 
 - Connected devices or roasting machine [e.g. Probatone 5]: 

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

CarefreeBuzzBuzz, 2021-06-17:
If you [look](https://artisan-scope.org/about/#platforms) you will see its not available for **native** iPad operation.  

Artisan supports the following 64bit operating systems:

macOS 10.15 and newer
Windows 8 and newer
Linux glibc 2.18
There are also experimental builds for the Raspberry Pi platform.

Something like the [Luna Display](https://astropad.com/product/lunadisplay/)  could work if you have a Mac and want to  run it on the iPad.   

Some have run it with  a  Raspberry Pi and and a tablet.   

Sorry, hope you find a way to do what you  need.

MakenuChocolate, 2021-06-17:
Thank you for your reply. If we went with the Luna Display could we use the
IPad to plug into the roaster or do you still need the labtop plugged into
the roaster and the iPad just turns into a second screen.

My main mission is to just have a tablet that plugs into the roaster that
we can track the roast. Is that possible on any tablet?

Thank you,
Tucker

CarefreeBuzzBuzz, 2021-06-17:
You haven't said what roaster it is or  what  devices it uses to read temperatures?  I use my MBA with a second monitor next to the roaster with an Apple Keyboard and Trackpad.  The  MBA sits on the bottom shelf of my roaster table.   If you wanted to avoid those the iPad with Luna Display could work but the Mac needs to be plugged into the roaster unless it's running a wireless protocol.  

You can read [Running Artisan on Raspberry Pi](https://artisan-roasterscope.blogspot.com/2019/02/running-artisan-on-raspberry-pi.html) for information on that route.   

There are  other [articles](https://artisan-scope.org/docs/documents/) that could be of interest as well.

MakenuChocolate, 2021-06-17:
The roaster is a Diedrich IR-5. We use the internal temperature in the
machine to read the temp.
What is MBA? I am sorry I am not very computer literate.

With the Luna Display I could not plug the IPad into the roaster?

Would a Windows Tablet work better?

Thank you again for all your help and support answering my questions

CarefreeBuzzBuzz, 2021-06-17:
MBA= Mac Book Air.   There is no plugging in a tablet.  You can plug in a computer unless your Diedrich supports wireless  set up.  You will have to look in  your manual  or  ask them.   With the Luna Display you  can run the full Artisan app on your iPad.  Please read their documentation as well because you may still need  the keyboard and mouse.   Sorry this isn't easier.   Might help if you post how the Diedrich connects.  I am moving this  to a discussion as its  really  not  an issue.

closed 2021-06-17 by CarefreeBuzzBuzz