artisan-roaster-scope/artisan #631: Feature request: add support for Decent Scale

decentjohn, 2021-06-18:
I see that you have support for the Acaia scale.

If you have any interest in adding support for my "Decent Scale"
https://decentespresso.com/scale

please let me know, as I'd be happy to send a free unit to who-ever wants to code it.

It's completely open, and I've documented the API 
https://decentespresso.com/decentscale_api

and am happy to assist any programmer working on it.

MarcelPr, 2021-06-18:
I am interested in coding and playing with the  Decent scale.

Vadim Olshevsky
5 Momin Dr, Millbury, MA 01527

https://www2.math.uconn.edu/~olshevsky/

decentjohn, 2021-06-18:
Great, thanks Vadim, I'll get one out to you.

If you have any questions, you can ask me here, or via email, as you prefer.

-john

decentjohn, 2021-06-18:
Hi Vadim,

To send you the Decent Scale, can I ask you that you tap BUY NOW here:
https://decentespresso.com/c?s=82+1

and then fill out your account info (name/pw) and address.  That way you have an account with us, get shipping tracking info, official tech support, etc…

do NOT fill out the credit card info on the last page.  

At that point, just msg me here and I'll approve the free scale, and get it sent off.

-john

MarcelPr, 2021-06-18:
Hi John,

Thank you so much! Great!
Order: 19625-30733

Vadim Olshevsky
https://www2.math.uconn.edu/~olshevsky/

decentjohn, 2021-06-18:
Thanks Vadim, scale will shortly be on its way.

MAKOMO, 2021-06-18:
Dear Marcel, I am looking forward for your first contribution to this project, or any other project here on Github! Hope you are not just grabbing a free scale and that way blocking progress on this project.

decentjohn, 2021-06-18:
@MAKOMO thanks for asking... I was assuming good faith. :D

MAKOMO, 2021-06-18:
@decentjohn Providing BLE support in cross-platform desktop app is tricky and requires quite some knowledge of the internals of Artisan and its underlying libs.  The support for the Acaia scale and integration in Artisan is quite complex. Does  Decent plans to  offer a scale supporting weights than 2kg, as required by most Artisan users, in the near future?

decentjohn, 2021-06-18:
It's super easy for us to make a 5kg scale in a few months. Will that be enough for Artisan's needs?   

> Providing BLE support in cross-platform desktop app is tricky and requires quite some knowledge of the internals of Artisan and its underlying libs. 

Sounds like a new contributor like @MarcelPr might not be up to that task. Do you have someone in mind who might be best for it?

As you already have Acaia support, the Decent Scale is not that different.  

We've got our own crossplatform code (using Tcl/Tk) that supports Acaia, Decent, and a few other scales, here's a link:
https://github.com/decentespresso/de1app/blob/7c8c0692f02fd5c0343d54c589f4ff91f61b2bb7/de1plus/bluetooth.tcl#L91

following that code to add Decent Scale support to the existing Acaia supporting code, shouldn't be that hard, since this documents what the equivalents are.

MarcelPr, 2021-06-18:
I overcommited myself for this summer and may not be immediately
contributing. So, if this free gift is for people who are able to start
right away, please send the scale to them.

Bestest.

decentjohn, 2021-06-18:
@MarcelPr I'd love to have you hack on the Decent Scale, but as this offer of a free scale was for Artisan support, it's probably not a good first project for you. However, if you do feel like playing with a scale, the $99 purchase option is there for you....

MarcelPr, 2021-06-18:
I did not read the discussion, just got an email from you offering a free
scale without any details about any project. So, I replied. Best.

decentjohn, 2021-06-18:
No worries @MarcelPr I think I should have been more precise in my initial FEATURE REQUEST above.