artisan-roaster-scope/artisan #639: About WebSockets Port

highmore, 2021-07-15:
Hello everyone, can I control all the functions of artisan through artisan's WebSockets port?
If the answer is yes, 
Can anyone provide detail of WebSockets interface information?

MAKOMO, 2021-07-15:
> 
> 
> Hello everyone, can I control all the functions of artisan through artisan's WebSockets port?

Not sure what exactly you refer to by "all the functions". The WebSocket interface is to READ data from a device and not to control Artisan.

> If the answer is yes,

The answer is most likely no.

> Can anyone provide detail of WebSockets interface information?

https://artisan-scope.org/devices/websockets/

M

highmore, 2021-07-15:
Oh, it is used to read data, and can't be used to send instructions?

MAKOMO, 2021-07-15:
> 
> 
> Oh, it is used to read data, and can't be used to send instructions?

Well, the WebSocket Command allows to send out JSON.

Check its documentation,
M

closed 2021-07-23 by MAKOMO